webusb/arduino #2: Support uploading sketches over WebUSB

reillyeon, 2016-02-03:
Once a sketch with support for WebUSB is uploaded it should be possible for updates to be uploaded over the WebUSB channel. This would allow embedded devices to be updated in the field by the same site that provides their user interface.

flaki, 2016-02-06:
How does the WebUSB process differs from other, JS-bound upload processes (like that of `[avrgirl-arduino](https://github.com/noopkat/avrgirl-arduino)`)?
That one uses `node-serial` to enter/restart in bootloader mode (opening the serialport at 1200 baud) and then uses the avr protocol to flash the chip.

@reillyeon could you please help me by elaborating on why the "_once a sketch with support for WebUSB is uploaded_" prerequisite is there? Thank you!

reillyeon, 2016-02-06:
The WebUSB library in this repo adds an additional interface on which WebUSB communication happens. A site using WebUSB will not have access to the serial interface on which the upload normally happens.

flaki, 2016-02-06:
Thanks for the answer Reilly, I did my fair reading on the spec on the meantime. So this "uploading" would practically happen through custom code uploaded to the device, capable of "self-modification", based on incoming data?

reillyeon, 2016-02-07:
No, the goal would be to upload a new image to the chip's normal flash memory through the bootloader. I have not yet done an investigation into what modifications to the bootloader will be necessary to make that possible.

chrissy25, 2018-05-06:
Hi, are there any updates on this request?

PaintedBlck, 2018-05-17:
I have found the following presentation of @noopkat : [I'm afraid your browser has been talking to the robots again a gentle intro to WebUSB](https://www.youtube.com/watch?v=ov2R3vEn14k) where she shows (27:30) that she can upload a sketch to an arduino. She also shows what tools she uses (29:47).

In short, she uses avr-girl variants. You can also find some inspiration on [noopkat/avrgirl-chrome-app](https://github.com/noopkat/avrgirl-chrome-app).

I hope this helps!

Grtz,
PaintedBlck

chrissy25, 2018-05-17:
Thanks for the hint :)

chrissy25, 2018-05-17:
Hm :( Are there any other plans to follow up on this @reillyeon?

PaintedBlck, 2018-05-17:
@chrissy25 I'm trying to do something in that direction. I've pushed the little bit of code I have made just now on [WebUSB-Arduino](https://github.com/PaintedBlck/WebUSB-Arduino). I'll try to work on it the next weeks.

I welcome all comments or useful hints.

reillyeon, 2018-05-17:
I think @PaintedBlck's confusion stems from the fact that there are many ways to program the application processor in an Arduino:

1. If the board uses a 32u4 or similar then the application processor is the only chip on the board and is responsible for running your sketch *and* communicating over the USB port. When the host sends the right signal over the USB port the chip puts itself in programming mode, receives the new firmware and burns it to flash.
2. If the board uses a 328P or similar then there is an additional chip between the application processor (running your sketch) and the USB port. On early Arduinos (and many other embedded devices) this is a USB-to-serial converter chip from a company like FTDI or Prolific. Modern Arduinos like the UNO use a 8u2 but the function is the same. The host drivers make it look like chip is a serial port with the application processor connected to it. The application processor never sees the USB traffic directly, only serial data on its TX/RX lines. By twiddling the serial port state the host can put the application processor into programming mode after which it will receive the new firmware over it's internally wired TX/RX pins and burn it to flash.
3. With either of these kinds of boards you can attach a "programmer" to a debug connector on the board. This is a completely separate interface specifically designed for programming and debugging (think JTAG). How the host communicates with the programmer varies. Sometimes it's a custom USB protocol. Sometimes it emulates a serial port.

What I was talking about in my earlier comments was adapting the firmware on the Arduino boards of type (1) so that they understand how to receive firmware from the USB interface that this library adds. Right now they only support receiving firmware on their first USB interface, which is a virtual serial port and inaccessible to the WebUSB API.

What @noopkat did was use a programmer as described in (3). I think the one they are using is one that speaks a custom USB protocol instead of presenting a virtual serial port because you can't connect to USB serial ports with WebUSB (the kernel driver claims the interface first).

That leaves the boards in configuration (2). To make these work you need to modify the firmware running on that chip between the USB port and the processor receiving and running your sketch. This modification would add an interface which is will not be claimed by the kernel USB serial port driver and so can be accessed by a page using WebUSB. Since on modern Arduino's that's just a 8u2 this is possible. If you look at an Arduino UNO you'll notice there are 2 3x2 jumper connectors on the board. One of those is the ISP (in-circuit programming) header for the application processor (what @noopkat used) and the other is the ISP header for the chip acting as USB-to-serial converter. Using this header and an appropriate programmer would allow you to change the firmware of that converter as much as necessary.

PaintedBlck, 2018-05-17:
@reillyeon Thank you for your clarification. Much appreciated!

q2dg, 2018-06-26:
For instance, you could add the ability of being a HID device (like Leonardo) to regular UNO boards with this specific firmware: https://github.com/NicoHood/HID

PsiOps, 2019-09-23:
@reillyeon thanks a lot for your analysis. Is there any progress on this functionality? I cannot seem to find a working implementation of this. 

To make sure I understand: So modifying the bootloader would have no effect on devices with a separate chip for usb<->serial? Why is that exactly? I have a clone Arduino device that uses a ch340 chip for this, and there is no header for programming that on the board. Does that make it impossible to flash over WebUSB?

jpliew, 2019-09-23:
@PsiOps board with dedicated USB to Serial chip like CH340, FTDI and etc will not be able to run WebUSB.

If you want flashing over WebUSB, check out https://github.com/devanlai/webdfu

reillyeon, 2019-10-03:
If you're looking for web-based flashing of devices over serial I recommend following @noopkat's work on porting noopkat/avrgirl-arduino to the [Serial API](https://wicg.github.io/serial).

noopkat, 2019-10-05:
current working branch of Serial API compatibility for avrgirl-arduino: https://github.com/noopkat/avrgirl-arduino/tree/chrome-serial

Currently Arduino Uno works, and Arduino Mega _almost_ works. Any 32u4 based Arduino board will bring some complications that I'm still thinking through and will almost certainly require more major refactoring on my side. But if you'd like to follow progress, the branch mentioned above is where it's at.