webusb/arduino #3: Out of stock boards

kenchris, 2016-02-04:
So the former is apparently being deprecated and the latter is out of stock everywhere.

How hard would it be getting something based on ATmega328 working, like the https://store.arduino.cc/product/GBX00066 ?

flaki, 2016-02-06:
To my understanding, the emphasis is on the chipset driving the Arduino - the ATmega32u4 has built-in inherent USB support, which is why it must have been chosen for these demos. You should be able to find plenty Arduino Micro clones out there on [Ebay](http://www.ebay.co.uk/sch/i.html?_from=R40&_trksid=p2050601.m570.l1313.TR0.TRC0.H0.Xatmega32u4.TRS0&_nkw=atmega32u4&_sacat=0)

reillyeon, 2016-02-06:
We're also looking into a new firmware for the ATmega16u2 found in boards using the ATmega328 as the application processor (such as the Arduino Uno R3). Right now the WebUSB library relies on the [PluggableUSB](https://github.com/arduino/Arduino/wiki/PluggableUSB-and-PluggableHID-howto) capability supported by the 32u4-based boards. The ones without built-in USB support have an additional chip that provides USB-to-serial conversion.

reillyeon, 2016-02-10:
It also appears that SparkFun has Arduino Leonardos [in stock](https://www.sparkfun.com/products/11286).

kenchris, 2016-02-11:
Great, I will try to acquire some

kenchris, 2016-02-11:
It seems that Arduino Yún is the logical replacement of the Leonardo, using the same processor:

https://www.arduino.cc/en/Guide/ArduinoYun

closed 2016-02-26 by reillyeon