webusb/arduino #4: Not finding devices

kenchris, 2016-02-25:
So I got a Leonardo and tried the console examples (adding the include and the #define Serial WebSerial)

Upload works fine and serial connection works fine without the WebSerial define.

I followed the description carefully here https://github.com/webusb/arduino and am on Dev on Windows but I never get any notification from chrome and when going to https://webusb.github.io/arduino/demos/console/ manually it says "No devices found"

kenchris, 2016-02-25:
Works fine with Linux

reillyeon, 2016-02-25:
The Microsoft OS Descriptors the device exposes are only honored by Windows 8.1 and above. If you're on Windows 7 it won't work unless you have an INF file instructing Windows to load winusb.sys. Sorry, I've been meaning to document exactly how the descriptors this code adds to the device work and thus this requirement.

kenchris, 2016-02-26:
I am on Windows 8.1 actually

reillyeon, 2016-02-26:
Can you verify in Device Manager that driver installation was successful and winusb.sys was successfully loaded? If that's all in place then this is likely a Chrome bug. Device enumeration on Windows is known to be flaky.

tongtongdlut, 2016-04-29:
I tried to use the demo rgb with a normal usb flash drive to try usb device detect. 
But when I clicked connect button,  there's a popup saying "no devices found". 
I'm using Mac OS, chrome Version 52.0.2716.0 dev (64-bit), the "Experimental Web Platform Features" flag is enabled. Is it a chrome bug or I missed some settings?

reillyeon, 2016-04-29:
The demo site expects you to have an Arduino with the correct firmware loaded. If you don't follow the instructions in the [README.md](https://github.com/webusb/arduino/blob/gh-pages/README.md#getting-started) it will not work.

trippedout, 2016-08-31:
Im using a [Flora board from Adafruit](https://learn.adafruit.com/getting-started-with-flora/overview), built on top of the Atmega32u4 chipset but can't seem to get the demo to work. Tried newest normal chrome and canary chrome, with WebUSB and Experimental flags enabled, uploads to board fine but never triggers browser to popup or when i manually visit site, says no devices connected.

On OSX with latest Arduino sdk (.11 so i shouldnt need to manually edit the USB header file anymore, right? i followed that issue chain on arduino's github) - any thoughts? Do i just need to edit the board file or am i missing something else?

reillyeon, 2016-09-01:
You will need to add a board configuration for the Flora similar to that included here for the Leonardo and Micro. It also looks like the new bootloader files are not included with the IDE so for the time being editing the header still looks like it is required.

ollietb, 2016-10-10:
@trippedout did you manage to sort out your issues? I seem to be having the same problem with Windows 10 & Arduino Micro

nagineni, 2016-11-17:
Issue reported by @kenchris is reproducible for me too. I tried with Arduino Leonardo on Windows 8.1 and 10, but no notification from Chrome. I can see the events 'Device configured (winusb.inf)' and 'Device started(WINUSB)' in the Device Manager.

cotejp, 2016-11-25:
I'm having the same issue on macOS Sierra with a Leonardo board. When I go to the [demo page](https://webusb.github.io/arduino/demos/console/), I get this error:

```
No devices found.
Connection error: NotFoundError: No device selected.
```
 
I tried with Chrome 54 and the latest Canary. The _Experimental Web Platform Features_ flag was  enabled in both.

cotejp, 2016-12-23:
Has anybody made progress on this? I tried with Arduino 1.8 and it still does not work. Is it working for anyone on Mac ?

mon, 2016-12-26:
This could be a driver problem. I've spent some time with Zadig playing with different combinations for both Interfaces. They're listed as interface 0 and 2, which AFAIK maps to a serial port (for programming, mainly) and WebUSB Console, respectively.

Forcing interface 2 to libusb made the Leonardo appear in the devices list (in Chrome's connect dialog) but it couldn't connect - `Failed to submit transfer: Input/Output Error` followed by `Failed to read WebUSB URL descriptor: ` Forcing them both to libusb made both appear in the list, but I didn't attempt to connect. I doubt it would have succeeded either, considering WinUSB is the recommended driver.

Under no circumstances could I get a WinUSB device to appear in the list, nor could I get the popup to appear to direct me to the demo page.

I'm not amazingly confident with descriptors, so take this with a grain of salt, but my own implementation's Control request handler for the my `bVendorCode` intentionally turns on a light and locks my board, and this never occurs. Something appears to be happening on the OS side to never request the information from the device.

oostap1, 2017-01-25:
@kenchris 
You mentioned that it works on linux. Any tricks except udev rules and --disable-webusb-security to make it work?
In my case (ubuntu 16.04) Chrome shows popup when device is connected but list of devices is empty. Even tried to enumerate all devices by navigator.usb.getDevices(), still nothing.

mon, 2017-01-25:
@oostap1 All I needed was udev, nothing else. Good luck on solving that one in your issue - everything looks fine at a quick glance.

oostap1, 2017-01-26:
@kenchris @mon 
Solved Ubuntu issue by reboot or dependencies Installation for chromium (not sure what helped).
As for now made it work on Mac and Linux.

Anyone, any luck with Windows? I cannot make it. Sometimes Device Manager even reports Failure to Get Descriptor for device.

@cotejp I have "Leonardo ETH" from some weird vendor "dog hunter AG" and vendor ID is different than specified in demo serial.js. Mine 0x2a03  (not 0x2341, seems it is clone) and as you can see it is not in a filter list. Check your vendor ID and product ID to make sure that they match demo. Or just get source, add own vendorID and productID and run local server "python -m SimpleHTTPServer" in demo folder (it did work on Mac).

cotejp, 2017-01-26:
@oostap1 I just tested it and it now works. I did not change anything. My unit is (and was) 0x2341/0x0836. The only obvious difference is that the test I just performed was with Chrome 57 (instead of 54 last time I tried). Anyway, thanks for your help.

oostap1, 2017-01-27:
Have just read WCID spec and it seems they require some kind of "special" string descriptor, called [Microsoft OS String Descriptor](https://github.com/pbatard/libwdi/wiki/WCID-Devices#Microsoft_OS_String_Descriptor). Is that descriptor still actual for Windows?

Hmm, I see that linux, as good boy, is requesting BOS descriptor after getting device descriptor. On Windows there is **NO** request for BOS and there **NO** request for mentioned 0xEE OS string descriptor. Not surprised that WebUSB does not work there. Not sure how Windows decided to load winusb driver after all...

mon, 2017-01-27:
@oostap1 that's the MS OS 1.0 descriptor and is deprecated in favour of MS OS 2.0. Are you certain your device is presenting itself as USB 2.1 and not 2.0?

oostap1, 2017-01-27:
@mon, Thank you for clarification. Yes it is exposed 2.1. lsusb on linux shows:

`Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x2a03 dog hunter AG
  idProduct          0x8042 
  bcdDevice            1.00
  iManufacturer           1 Unknown
  iProduct                2 Arduino Leonardo ETH
  iSerial                 3 WUART
  bNumConfigurations      1`

I tried even set to be 3.0 in USB_VERSION. No luck. Do you have it working on Windows? If yes, could you share Win version?

reillyeon, 2017-01-27:
From the MSDN documentation:

> After the operating system requests a Microsoft OS String Descriptor from a device, it creates the following registry key:
>
> **HLKM\SYSTEM\CurrentControlSet\Control\UsbFlags\vvvvpppprrrrr**
>
> The operating system creates a registry entry, named osvc, under this registry key that indicates whether the device supports Microsoft OS Descriptors. If the device does not provide a valid response the first time that the operating system queries it for a Microsoft OS String Descriptor, the operating system will make no further requests for that descriptor.

"Uninstall" the device from the Device Manager and clear that registry key to make Windows try again.

oostap1, 2017-01-27:
@reillyeon I did all what you sugested already and even tried to fool MS by creation such registry. Nothing. As @mon mentioned it is MS OS 1.0 descriptor and replaced by MS OS 2.0, which is implemented in WebUSB.cpp.

Finnaly got external USB sniffer and discovered that wireshark just does not show everything :( on windows (some requests are missing). Maybe I need the same version as on linux or Win does not allow to read them.

Anyway I have two USB dumps now form MS and linux.

They are "adequately similar", some difference in order of requests,t I do not believe it is a case.
[image]
Windows, indeed, retrieves everything from device. What is missing is Chrome communication with device. On Left (linux) of screenshoot there are GetDescriptor(BOS)->Vendor Request(s) (0x01) to get URLs. That part is missing on Windows. @reillyeon  any idea how to get more debug info from Chrome?

oostap1, 2017-01-27:
Chrome logs [how to enable](https://www.chromium.org/for-testers/enable-logging) shows weird failures:

> [7288:10172:0127/113109.119:VERBOSE1:device_event_log_impl.cc(142)] [11:31:09.119] USB: EVENT: usb_device_impl.cc:76 Failed to get config descriptor: Entity not found
[7288:10172:0127/113109.119:VERBOSE1:device_event_log_impl.cc(142)] [11:31:09.119] USB: EVENT: usb_device_impl.cc:96 Failed to get active configuration: Entity not found
[7288:10172:0127/113109.119:VERBOSE1:device_event_log_impl.cc(142)] [11:31:09.119] USB: USER: usb_service_impl.cc:478 USB device added: vendor=32902 "", product=39985 "", serial="", guid=d8ed2d82-2369-4729-be7a-e3216d81c0ce
[7288:10172:0127/113109.119:VERBOSE1:device_event_log_impl.cc(142)] [11:31:09.119] USB: EVENT: usb_device_impl.cc:76 Failed to get config descriptor: Entity not found
[7288:10172:0127/113109.119:VERBOSE1:device_event_log_impl.cc(142)] [11:31:09.119] USB: EVENT: usb_device_impl.cc:96 Failed to get active configuration: Entity not found
[7288:10172:0127/113109.119:VERBOSE1:device_event_log_impl.cc(142)] [11:31:09.119] USB: USER: usb_service_impl.cc:478 USB device added: vendor=32902 "", product=39974 "", serial="", guid=f4cc7a7b-fb68-448d-b977-ed7534a9806b
[7288:10172:0127/113109.119:VERBOSE1:device_event_log_impl.cc(142)] [11:31:09.119] USB: USER: usb_service_impl.cc:478 USB device added: vendor=32903 "", product=2012 "", serial="", guid=34a5413b-3af8-4f03-99e8-9ac88dfbcc0e
[7288:10172:0127/113109.119:VERBOSE1:device_event_log_impl.cc(142)] [11:31:09.119] USB: EVENT: usb_device_impl.cc:111 Failed to open device: Operation not supported or unimplemented on this platform
[7288:10172:0127/113109.120:VERBOSE1:device_event_log_impl.cc(142)] [11:31:09.119] USB: EVENT: usb_device_impl.cc:111 Failed to open device: Entity not found
[7288:10172:0127/113109.120:VERBOSE1:device_event_log_impl.cc(142)] [11:31:09.120] USB: EVENT: usb_device_impl.cc:111 Failed to open device: Operation not supported or unimplemented on this platform
[7288:10172:0127/113109.120:VERBOSE1:device_event_log_impl.cc(142)] [11:31:09.120] USB: EVENT: usb_device_impl.cc:111 Failed to open device: Entity not found
[7288:10172:0127/113109.121:VERBOSE1:device_event_log_impl.cc(142)] [11:31:09.120] USB: USER: usb_service_impl.cc:478 USB device added: vendor=1133 "Logitech", product=50479 "USB Receiver", serial="", guid=f1bcb462-a38f-4f08-954b-ad8a99a8b188

oostap1, 2017-01-30:
@reillyeon @mon
Finally was able to get rid of Zadig. More details in pull [request](https://github.com/webusb/arduino/pull/28) and webusb github [issue](https://github.com/WICG/webusb/issues/75). Still **NO** landing page popup.

DeqingSun, 2017-02-05:
Hi, I tried win7 computer with my inf file. But Chrome don't recognize the hardware.
Is there an INF example?

Thanks

foxxyz, 2017-06-30:
Still no luck on Windows 10 using Arduino Micro with example `console` sketch loaded on it and USB 2.1. Works like a charm on MacOS/Linux.

@oostap1 Tried to follow your instructions and added WinUSB driver (6.1) via Zadig. Chrome sees the device now, but when trying to connect the webpage reports "Unable to claim interface".

ynot729, 2017-10-15:
I was also able to make it show up in the pop up for Windows 10 using zadig. But unable to claim the interface. Has anyone been able to get the RGB demo to work on Windows 10?

alexandr-, 2017-10-27:
@ynot729  have you tried ooostap1 solution? worked for me!
-> https://github.com/webusb/arduino/pull/28

brianz, 2017-11-02:
I managed to get the demo working on OSX using a board from Adafruit, the Feather 32u4.  Hopefully, this is useful for others trying to work with a different board.

Here's what I did:

- Updated the build flags in the `boards.txt` file, which is located in `~/Library/Arduino1/packages/adafruit/hardware/avr/1.4.9/`.  I assumed this was needed since [boards.txt in this repo](https://github.com/webusb/arduino/blob/gh-pages/hardware/webusb/avr/boards.txt#L28) has these settings.

```
feather32u4.build.extra_flags={build.usb_flags} "-DUSB_VERSION=0x210"
```

- When editing this file, take note of the `vid` and `pid` which identify the board over USB.

```
 feather32u4.build.vid=0x239A
 feather32u4.build.pid=0x800C
```

- Added the Feather's `vid` and `pid` in the list of filters in `serial.js`.  These two values were pulled from the `boards.txt` file above:

```
--- a/demos/serial.js
+++ b/demos/serial.js
@@ -17,6 +17,7 @@ var serial = {};
       { 'vendorId': 0x2341, 'productId': 0x804e },
       { 'vendorId': 0x2341, 'productId': 0x804f },
       { 'vendorId': 0x2341, 'productId': 0x8050 },
+      { 'vendorId': 0x239A, 'productId': 0x800C },
```

- Ran the demo locally, simply serving static files from this repo's `demos` directory on `localhost:8888`.
- Updated the sketch to point to `localhost`:

```
WebUSB WebUSBSerial(1, "localhost:8888");
```

After uploading, the notification pops up and I'm able to pair the device.

basevers, 2018-01-24:
Just got my first Arduino board, I got it specifically to experiment with WebUSB.

After one day of fiddling around, here are my "Out Of The Box" observations:
Using: MKRZERO, Windows 10, Arduino 1.8.5, SAMD tool chain 1.6.17, Chrome 63

1) Importing the WebUSB library caused issues as it seems the WebUSB directory structure isn't how the IDE would like to see it.
Its an easy change by creating a file "README.adoc" and creating a subdirectory "src" for the WebUSB.cpp and WebUSB.h files.  (you can see how it's done in the other libraries) 

2) Compile error (only for RGB sketch)
_int index;_ 
Did cause a compile error as there is something wrong with the tool chain.
You can simply renamed the variable "_index_" to "_colorIndex_"

3) Compile warnings (multiple)
Redefine "_Serial_" as it is already defined in: Arduino15\packages\arduino\hardware\samd\1.6.17\variants\mkrzero/variant.h:196:0:
Plus a few other warnings, for now I ignored them, but I think its part of the other issues I'm facing.

4) Code hangs in _setup()_; waiting for "_Serial_"
Commenting out _//#define Serial WebUSBSerial_
Resulted seeing ""Sketch begins."
This seems there is an issue when WebUSB is plugged as a node to the existing USB descriptor

5) USB_VER 0x0210
For the MKRZERO its needs to be changed in
_#define D_DEVICE(_class,_subClass,_proto,_packetSize0,_vid,_pid,_version,_im,_ip,_is,_configs) \
	{ 18, 1, 0x200, _class,_subClass,_proto,_packetSize0,_vid,_pid,_version,_im,_ip,_is,_configs }_
However, changing 0x200 to 0x210, made Windows 10 no longer recognize the board. Resulting that the board couldn't be reprogrammed. I had to force the board into the bootloader by pressing the reset button quickly two times.

Conclusion: WebUSB is still not working for me.

My objective is to use a https: webpage to communicate to a (I2C or SPI) peripheral. for example: I would like a use an temperature and humidity sensors displaying real-time data on a webpage.

Maybe the quickest moving forward is to use a Arduino board that seems to work "Out Of The Box"

Please feel free to make suggestions.

reillyeon, 2018-01-24:
@basevars, can you create separate issues for these problems? A number of these may be specific to using the WebUSB library on the SAMD architecture which I unfortunately don't have the hardware on hand to test.

basevers, 2018-01-26:
Thanks for the fast response, ill split them up

pierrejoye, 2019-07-07:
old post but surely useful for other

using https://zadig.akeo.ie/ you can force windows to use winsub driver for a given device.  Amazingly handy instead of own inf file and all these cumbersome steps.