webusb/arduino #6: Why USB 2.1?

beaufortfrancois, 2016-02-26:
Just a quick question there: why do we require USB 2.1?
Is this a real thing?
If so, is that even considerable to ask arduino folks to update their file? https://github.com/arduino/Arduino/blob/06e9981e6b8f302552280e55ce3adc81b0bcc2ee/hardware/arduino/avr/cores/arduino/USBCore.h#L263

reillyeon, 2016-02-26:
USB 2.1 is required so that the host knows that the device exposes a Binary Object Store descriptor, which is where the WebUSB descriptor (and Microsoft OS Descriptor 2.0) reading sequence starts.

beaufortfrancois, 2016-02-29:
Thank you @reillyeon.
I've sent a PR at https://github.com/arduino/Arduino/pull/4627

Let's cross our fingers ;)

beaufortfrancois, 2016-03-09:
Just got an answer from Arduino folks who told me the patch will be merged in the next core release.
Francois happy!

reillyeon, 2017-02-16:
Cleaning up old issues.

closed 2017-02-16 by reillyeon