webusb/arduino #9: can only find devices from https://webusb.github.io

monteslu, 2016-03-04:
When I host the same files on another HTTPS site, it wont list devices.  Works fine from the github pages site.

Also, when plugging in a Micro or Leonardo I get a notification to go to the github pages site.  Where is that coming from?

Version 51.0.2666.0 canary (64-bit)  Mac OS 10.10.5

sowbug, 2016-03-04:
See https://github.com/WICG/webusb/blob/gh-pages/explainer.md for an
overview. In particular "USB devices will be associated with a web origin
and can only be connected to from a page from the same origin." The landing
page descriptor, which lives in the device firmware, determines the content
of the notification.

monteslu, 2016-03-04:
Would it make sense to allow WebUSB.cpp to be generic and have us pass in the whitelisted URLs from our .ino files?

reillyeon, 2016-03-04:
Duplicate of issue #1.

closed 2016-03-04 by reillyeon