webusb/arduino #13: How to make work with other ATmega32u4 devices?

tdicola, 2016-04-28:
Hello, sorry that this is more of a support request and not an issue but I'm having a lot of trouble figuring out how to make the demos work with an ATmega32u4-based Arduino-compatible board.  Everything works great with an official Arduino Leonardo, but I'm trying to use a board with the same CPU (Circuit Playground, https://www.adafruit.com/product/3000) and it's not showing up in the list of USB devices when you click the connect button in the console demo.  Chrome (canary build from today) does pop up a notification that a web USB device was connected and recommends the URL to visit, but going to the page and clicking connect shows no device unfortunately.

My suspicion is that somewhere the VID & PID of the Arduino Leonardo are hard coded in--does that sound right?  This Circuit Playground board has a different VID & PID (VID = 0x239a, PID = 0x8011) so it would make sense that something expecting to find a Leonardo wouldn't catch it.  However I don't see any obvious place where there's a dependency on the VID & PID in either the sketch or demo web page code.

One thing I am curious about, what exactly is encoded in this information in the console index.html origin-trial headers?  https://github.com/webusb/arduino/blob/gh-pages/demos/console/index.html#L4-L7  Is there a VID & PID embedded in that presumably base64 encoded data?  Also is the format of this blob documented somewhere in the draft web USB spec?  I searched around it but just came away even more confused--there doesn't seem to be any mention of the origin-trial header and this data (but I'm not really familiar with W3C specs like this so it could just be me missing something obvious).

In general though do the demos right now have a hard dependency on the Arduino Leonardo's VID & PID?  And if so can you point out what to change to make the demos work with a board that has a different VID & PID (but the same ATmega32u4 processor)?  Thanks!

reillyeon, 2016-04-28:
If you are using different hardware you should add the IDs to list at [serial.js:13](https://github.com/webusb/arduino/blob/gh-pages/demos/serial.js#L13).

The `<meta http-equiv="origin-trial">` tag is for Chrome's [Origin Trials](https://github.com/jpchase/OriginTrials/blob/gh-pages/explainer.md) framework under which WebUSB will eventually launch as an experimental API. It is not part of the WebUSB spec.

tdicola, 2016-04-28:
Oh ack somehow missed that when searching.  Perfect, thanks!

closed 2016-04-28 by tdicola