webusb/arduino #18: Compile time errors via WebUSB

samccone, 2016-07-16:
```
Arduino: 1.6.9 (Mac OS X), Board: "Arduino/Genuino Uno"

In file included from /Users/sam/Desktop/repos/arduino/demos/rgb/sketch/sketch.ino:1:0:
/Users/sam/Downloads/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/WebUSB/WebUSB.h:36:2: error: 'InterfaceDescriptor' does not name a type
  InterfaceDescriptor dif;
  ^
/Users/sam/Downloads/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/WebUSB/WebUSB.h:37:2: error: 'EndpointDescriptor' does not name a type
  EndpointDescriptor  in;
  ^
/Users/sam/Downloads/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/WebUSB/WebUSB.h:38:2: error: 'EndpointDescriptor' does not name a type
  EndpointDescriptor  out;
  ^
/Users/sam/Downloads/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/WebUSB/WebUSB.h:47:41: error: expected class-name before ',' token
 class WebUSB : public PluggableUSBModule, public Stream
                                         ^
/Users/sam/Downloads/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/WebUSB/WebUSB.h:68:27: error: 'SERIAL_BUFFER_SIZE' was not declared in this scope
  unsigned char _rx_buffer[SERIAL_BUFFER_SIZE];
                           ^
/Users/sam/Downloads/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/WebUSB/WebUSB.h:94:20: error: 'USBSetup' has not been declared
  int getDescriptor(USBSetup& setup);
                    ^
/Users/sam/Downloads/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/WebUSB/WebUSB.h:95:13: error: 'USBSetup' has not been declared
  bool setup(USBSetup& setup);
             ^
/Users/sam/Downloads/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/WebUSB/WebUSB.h:99:28: error: 'USBSetup' has not been declared
  bool VendorControlRequest(USBSetup& setup);
                            ^
exit status 1
Error compiling for board Arduino/Genuino Uno.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

samccone, 2016-07-16:
closing as WAI -- Uno is not supported due to The pluggable USB API not being available

closed 2016-07-16 by samccone

astagi, 2016-10-04:
@samccone this could help https://github.com/NicoHood/HoodLoader2