webusb/arduino #21: JS errors whilst running 'console' sample

luukau, 2016-10-09:
Aiming to remote console and program (using avrdude) an Arduino derivative board via WebUSB, I came across this as a good start. Having followed steps 1 and 2 from the 'getting started'. Whilst launching the console (ie http://webusb.github.io/arduino/demos/console/), I noticed the following JS errors appear in chrome:

"Uncaught TypeError: Cannot read property 'getDevices' of undefined
    at Object.serial.getPorts (serial.js:7)
    at HTMLDocument.document.addEventListener.event (console.js:72)"

and when I subsequently click "Connect':

"serial.js:17 Uncaught TypeError: Cannot read property 'requestDevice' of undefined
    at Object.serial.requestPort (serial.js:17)
    at HTMLButtonElement.<anonymous> (console.js:63)"

These error(s) seem to hint that 'navigator.usb.getDevices' and 'navigator.usb.requestDevice' are not supported in this version of chrome. I grabbed the latest chrome from the dev channel, as per step 1 (ended up with v55.0.2882.0) and ensured the correct flags were enabled.

My question; do you think this is a chrome issue or would it be because I do not (yet) have a board connected via USB? Not ruling out that there is something else I am missing either.

I currently can't connect my board as I 'bricked' it.. doh. I thought I'd ask the question regarding these JS error in chrome, whilst the board is being fixed.

reillyeon, 2016-10-10:
The WebUSB API is only available on pages loaded over HTTPS, to prevent this powerful feature from being abused by code injected into insecure HTTP connections. GitHub Pages has a nasty habit of redirecting from HTTPS to HTTP whenever you enter a URL without a trailing slash which is probably how you ended up there.

closed 2016-10-10 by reillyeon

luukau, 2016-10-10:
Sweet, thanks Reilly that certainly did the trick. Oversight on my behalf.