webusb/arduino #26: WebUSB.h:36:2: error: 'InterfaceDescriptor' does not name a type

b45ch1, 2017-01-04:
I followed the instructions from https://github.com/webusb/arduino/blob/gh-pages/README.md to build /arduino/demos/rgb/sketch/sketch.ino. However, the build fails. I'm on Ubuntu 14.04 using Arduino 1.8.0.

Below you can find the complete build log. It appears that InterfaceDescriptor, EndpointDescriptor, etc. cannot be found. Am I missing a dependency?

```
/home/swalter/software/arduino-1.8.0/arduino-builder -dump-prefs -logger=machine -hardware /home/swalter/software/arduino-1.8.0/hardware -tools /home/swalter/software/arduino-1.8.0/tools-builder -tools /home/swalter/software/arduino-1.8.0/hardware/tools/avr -built-in-libraries /home/swalter/software/arduino-1.8.0/libraries -libraries /home/swalter/Arduino/libraries -fqbn=arduino:avr:uno -vid-pid=0X2341_0X0043 -ide-version=10800 -build-path /tmp/arduino_build_575522 -warnings=default -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/home/swalter/software/arduino-1.8.0/hardware/tools/avr -prefs=runtime.tools.avr-gcc.path=/home/swalter/software/arduino-1.8.0/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/home/swalter/software/arduino-1.8.0/hardware/tools/avr -verbose /home/swalter/tmp/arduino/demos/rgb/sketch/sketch.ino
/home/swalter/software/arduino-1.8.0/arduino-builder -compile -logger=machine -hardware /home/swalter/software/arduino-1.8.0/hardware -tools /home/swalter/software/arduino-1.8.0/tools-builder -tools /home/swalter/software/arduino-1.8.0/hardware/tools/avr -built-in-libraries /home/swalter/software/arduino-1.8.0/libraries -libraries /home/swalter/Arduino/libraries -fqbn=arduino:avr:uno -vid-pid=0X2341_0X0043 -ide-version=10800 -build-path /tmp/arduino_build_575522 -warnings=default -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/home/swalter/software/arduino-1.8.0/hardware/tools/avr -prefs=runtime.tools.avr-gcc.path=/home/swalter/software/arduino-1.8.0/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/home/swalter/software/arduino-1.8.0/hardware/tools/avr -verbose /home/swalter/tmp/arduino/demos/rgb/sketch/sketch.ino
Using board 'uno' from platform in folder: /home/swalter/software/arduino-1.8.0/hardware/arduino/avr
Using core 'arduino' from platform in folder: /home/swalter/software/arduino-1.8.0/hardware/arduino/avr
Detecting libraries used...
"/home/swalter/software/arduino-1.8.0/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10800 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/home/swalter/software/arduino-1.8.0/hardware/arduino/avr/cores/arduino" "-I/home/swalter/software/arduino-1.8.0/hardware/arduino/avr/variants/standard" "/tmp/arduino_build_575522/sketch/sketch.ino.cpp" -o "/dev/null"
"/home/swalter/software/arduino-1.8.0/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10800 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/home/swalter/software/arduino-1.8.0/hardware/arduino/avr/cores/arduino" "-I/home/swalter/software/arduino-1.8.0/hardware/arduino/avr/variants/standard" "-I/home/swalter/software/arduino-1.8.0/libraries/WebUSB" "/tmp/arduino_build_575522/sketch/sketch.ino.cpp" -o "/dev/null"
"/home/swalter/software/arduino-1.8.0/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10800 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/home/swalter/software/arduino-1.8.0/hardware/arduino/avr/cores/arduino" "-I/home/swalter/software/arduino-1.8.0/hardware/arduino/avr/variants/standard" "-I/home/swalter/software/arduino-1.8.0/libraries/WebUSB" "/home/swalter/software/arduino-1.8.0/libraries/WebUSB/WebUSB.cpp" -o "/dev/null"
Generating function prototypes...
"/home/swalter/software/arduino-1.8.0/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10800 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/home/swalter/software/arduino-1.8.0/hardware/arduino/avr/cores/arduino" "-I/home/swalter/software/arduino-1.8.0/hardware/arduino/avr/variants/standard" "-I/home/swalter/software/arduino-1.8.0/libraries/WebUSB" "/tmp/arduino_build_575522/sketch/sketch.ino.cpp" -o "/tmp/arduino_build_575522/preproc/ctags_target_for_gcc_minus_e.cpp"
"/home/swalter/software/arduino-1.8.0/tools-builder/ctags/5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "/tmp/arduino_build_575522/preproc/ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"/home/swalter/software/arduino-1.8.0/hardware/tools/avr/bin/avr-g++" -c -g -Os  -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10800 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/home/swalter/software/arduino-1.8.0/hardware/arduino/avr/cores/arduino" "-I/home/swalter/software/arduino-1.8.0/hardware/arduino/avr/variants/standard" "-I/home/swalter/software/arduino-1.8.0/libraries/WebUSB" "/tmp/arduino_build_575522/sketch/sketch.ino.cpp" -o "/tmp/arduino_build_575522/sketch/sketch.ino.cpp.o"
In file included from /home/swalter/tmp/arduino/demos/rgb/sketch/sketch.ino:1:0:
/home/swalter/software/arduino-1.8.0/libraries/WebUSB/WebUSB.h:36:2: error: 'InterfaceDescriptor' does not name a type
  InterfaceDescriptor dif;
  ^
/home/swalter/software/arduino-1.8.0/libraries/WebUSB/WebUSB.h:37:2: error: 'EndpointDescriptor' does not name a type
  EndpointDescriptor  in;
  ^
/home/swalter/software/arduino-1.8.0/libraries/WebUSB/WebUSB.h:38:2: error: 'EndpointDescriptor' does not name a type
  EndpointDescriptor  out;
  ^
/home/swalter/software/arduino-1.8.0/libraries/WebUSB/WebUSB.h:47:41: error: expected class-name before ',' token
 class WebUSB : public PluggableUSBModule, public Stream
                                         ^
/home/swalter/software/arduino-1.8.0/libraries/WebUSB/WebUSB.h:68:27: error: 'SERIAL_BUFFER_SIZE' was not declared in this scope
  unsigned char _rx_buffer[SERIAL_BUFFER_SIZE];
                           ^
/home/swalter/software/arduino-1.8.0/libraries/WebUSB/WebUSB.h:94:20: error: 'USBSetup' has not been declared
  int getDescriptor(USBSetup& setup);
                    ^
/home/swalter/software/arduino-1.8.0/libraries/WebUSB/WebUSB.h:95:13: error: 'USBSetup' has not been declared
  bool setup(USBSetup& setup);
             ^
/home/swalter/software/arduino-1.8.0/libraries/WebUSB/WebUSB.h:99:28: error: 'USBSetup' has not been declared
  bool VendorControlRequest(USBSetup& setup);
                            ^
Using library WebUSB at version 1.0 in folder: /home/swalter/software/arduino-1.8.0/libraries/WebUSB 
exit status 1
Error compiling for board Arduino/Genuino Uno.
```

reillyeon, 2017-01-04:
You are attempting to build for Arduino/Genuino Uno which does not support Pluggable USB.

reillyeon, 2017-01-04:
I've updated WebUSB.h to display a better error when the board configuration is unsupported.

closed 2017-01-04 by reillyeon

b45ch1, 2017-01-12:
Thanks. I must have missed the hardware requirements. I ordered an Arduino micro and succeeded to run the WebUSB demos on Ubuntu 14.04, Chromium 53.0.2785.143 after adding
```
SUBSYSTEM=="usb",ATTR{idVendor}=="2341",ATTR{idProduct}=="8037", MODE:="0666", GROUP:="plugdev"
```
to `/etc/udev/rules.d/50-arduino-micro.rules`.