webusb/arduino #29: SecurityError loading RGB demo with Circuit Playground

islemaster, 2017-05-05:
I'm attempting to get the RGB demo up and running with an Adafruit Circuit Playground board. I've already added Circuit Playground's VID and PID to [serial.js](https://github.com/webusb/arduino/blob/gh-pages/demos/serial.js#L13) as mentioned in https://github.com/webusb/arduino/issues/13.  When I click the "Connect" button I can select my device, but then the error handler prints `SecurityError: Access denied.`

The only time `SecurityError` is mentioned in the WebUSB spec is in [Section 4 - Device Enumeration](https://wicg.github.io/webusb/#enumeration) under the section starting 'To request the "usb" permission...'

> If the algorithm is not triggered by user activation, reject promise with a SecurityError and abort these steps.

I'm definitely interacting when this error occurs, so I'm not sure where my setup is going wrong.

- Attempting at commit ff4aae6fc7c4a332a410f8329f0a386bda44ac5f
- On Ubuntu 17.04 64-bit
- Using Google Chrome 58.0.3029.96 (Official Build) (64-bit) with flags `--enable-experimental-web-platform-features` and `--enable-features=WebUSB`
- I've followed the README instructions, including editing `USBCore.h` to enable USB 2.1.
- Uploaded [sketch.ino](https://github.com/webusb/arduino/blob/ff4aae6fc7c4a332a410f8329f0a386bda44ac5f/demos/rgb/sketch/sketch.ino)  to the Circuit Playground with no customizations, using Arduino IDE 1.8.1.
- Using [http-server](https://github.com/indexzero/http-server) to run the demo page at http://localhost:8000/rgb/index.html
- Maybe notable: I don't get a notification from Chrome when I plug in the Circuit Playground.

Any ideas?  Thanks!

reillyeon, 2017-05-05:
There's a missing step in the setup instructions. On Linux you need to add a file containing the following line to `/etc/udev/rules.d` to tell the OS to set permissions on the device so that Chrome can open it,

```
SUBSYSTEMS=="usb", ATTR{idVendor}=="your VID here", ATTR{idProduct}=="your PID here", MODE:="0666"
```

I'm going to re-purpose this bug for documenting this in README.md.

islemaster, 2017-05-05:
Interesting... I already had the following in `/etc/udev/rules.d/99-adafruit-boards.rules`, although it looks malformed compared to the rule you posted:
```
# Rule to make Trinket/Pro Trinket/Gemma/Flora programmable without running Arduino as root.
# Tested with Ubuntu 14.04 and 12.04.  Other distributions might need to update GROUP="dialout"
# to another group value like "users".
SUBSYSTEM=="usb", ATTR{idProduct}=="0c9f", ATTRS{idVendor}=="1781", MODE="0660", GROUP="dialout"

# Rule to blacklist Adafruit USB CDC boards from being manipulated by ModemManager.
# Fixes issue with hanging references to /dev/ttyACM* devices on Ubuntu 15.04.
ATTRS{idVendor}=="239a", ENV{ID_MM_DEVICE_IGNORE}="1"
```

Adding the rule you provided with correct Circuit Playground VID/PIDs swapped in...
```
SUBSYSTEMS=="usb", ATTR{idVendor}=="239a", ATTR{idProduct}=="8011", MODE:="0666"
```
...definitely gets me further than before.  I still don't get a notification from Chrome on plug, but the RGB demo now gets as far as "Connected" and seems to _think_ it's setting LED colors when I move the sliders.  Unfortunately nothing seems to actually happen on the board.  I'll keep experimenting and post here if I can work out how to control the LEDs.

reillyeon, 2017-05-05:
If you look in chrome://device-log when you plug in your device there should be a log message explaining why it failed to read the landing page URL.

superquest, 2018-08-27:
I'm having the same problem and adding this line doesn't have any effect.

The output of `udevadm info -a -n /dev/ttyACM0` is

```
  looking at device '/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/tty/ttyACM0':
    KERNEL=="ttyACM0"
    SUBSYSTEM=="tty"
    DRIVER==""
...
```

It seems that my Arduino isn't a member of the "usb subsystem"? 

I'm stuck ...

superquest, 2018-08-27:
I was able to get it to run by following these instructions: https://learn.adafruit.com/adafruit-arduino-ide-setup/linux-setup

xylo04, 2020-06-29:
I had a very smooth experience today using only `README.md` to get running with Linux, Chrome 83 and an Adafruit M0, with the exception of having to make a `udev` rule. What surprises me is that Arduino IDE is able to program and use the serial console just fine without the `udev` rule, but Chrome was just giving me the cryptic "SecurityError: Access denied" message.

I added `/etc/udev/rules.d/51-adafruit.rules`: 

```
SUBSYSTEM=="usb", ATTR{idVendor}=="239a", GROUP="plugdev"
```

My OS user was already part of the `plugdev` group.

gauntface, 2022-06-18:
Are you open to taking Pull Requests to add this to the README?

reillyeon, 2022-06-21:
Yes.