webusb/arduino #30: Update code to match changes in the spec

prusnak, 2017-05-08:
I am talking mainly about changes in the Feature Policy here: https://github.com/WICG/webusb/commit/c9646137abd41606982c5cf2703ac42629e34616 (removal of GET_ALLOWED_ORIGINS, etc.)

But maybe some other changes in the spec are relevant for this example (which is a go-to point for reference code of all hardware vendors implementing the spec).

reillyeon, 2017-05-11:
I have a patch which does this however due to [a bug](https://bugs.chromium.org/p/chromium/issues/detail?id=721546) in Chrome I will be waiting a little bit before merging it.

prusnak, 2017-05-12:
I am confused. You are talking about removal of GET_ALLOWED_ORIGINS, https://github.com/WICG/webusb/commit/c9646137abd41606982c5cf2703ac42629e34616 also talks about the removal, yet [this comment](https://bugs.chromium.org/p/chromium/issues/detail?id=721546#c1) says it is optional. What is the state of GET_ALLOWED_ORIGINS?

reillyeon, 2017-05-12:
Before WICG/webusb@c964613 GET_ALLOWED_ORIGINS was required by the spec and in Chrome. In Chrome 57 and 58 it started to be ignored when making permission decisions for top-level frames however due to [this issue](https://bugs.chromium.org/p/chromium/issues/detail?id=721546) it wasn't completely optional because without it Chrome wouldn't bother reading the landing page URL. Now that that issue is fixed and will be rolling out in Chrome 59 devices truly don't have to implement it anymore. The optional state was a decision within Chrome that was not reflected in the specification. I am currently landing [a patch](https://codereview.chromium.org/2815003005) that brings Chrome up to date with the specification.

prusnak, 2017-11-28:
Is BOS descriptor still required? If not, can we please further reduce the example not to include it?

reillyeon, 2017-11-28:
The BOS descriptor is optional. We will continue to include both the WebUSB and Microsoft BOS capability descriptors in this example as they provide an improved user experience by informing the browser of a landing page for the device and informing Windows that the winusb.sys driver should be loaded.

closed 2017-11-28 by reillyeon