webusb/arduino #32: USB_VERSION 0x200 and change 0x200 to 0x210 not needed

jpliew, 2017-05-21:
> The WebUSB library provides all the extra low-level USB code necessary for WebUSB support except for one thing: Your device must be upgraded from USB 2.0 to USB 2.1. To do this go into the SDK installation directory and open hardware/arduino/avr/cores/arduino/USBCore.h. Then find the line #define v. That's it!

This change is not needed because the file webusb\avr\boards.txt in this repo has already defined the correct USB_VERSION.

Just need to add a missing instruction in README.md to copy the folder webusb to arduino\hardware folder.

q2dg, 2017-08-08:
Maybe adding a new "WebUsb Board" into official Arduino Boards Manager could do this trick automatically?

jpliew, 2017-12-20:
@q2dg when the `webusb` folder from this repo is being copied to Arduino's `hardware` folder, the IDE will have WebUSB shown in the tools-->board section

[image]

Editing the original  `USBCore.h` will interfere with other boards that is relying on the original code.

jpliew, 2018-02-10:
Hi @reillyeon I wonder why you change the title of this issue to "Officially support using the WebUSB versions of board definitions"?

Officially, this library WebUSB, already support board definitions. 

There is just a missing instruction to copy the files over to Arduino and does not require editing of the `USBcore.h` file.

So, this issue is to correct the README.md. If you are happy I can do a pull request to change the wording.

reillyeon, 2018-02-11:
The necessary action on this issue is to verify that the updated board definitions correctly set USB_VERSION for all the boards listed as tested, and then update README.md.