webusb/arduino #34: Controlling stepper motor via DRV8825

c0debreaker, 2017-06-27:
I have a DRV8825 stepper driver. I've used it with Arduino UNO. Can webusb control drv8825?

dataino-it, 2017-06-27:
no is not possible

see my question on arduino forum
https://forum.arduino.cc/index.php?topic=464751.0

c0debreaker, 2017-06-27:
I forgot to say that I want to use Arduino 101. When I went to Fluent Conf last week, the person told me to buy Arduino 101. He showed a demo and I saw his javascript code. He turned on the built-in LED, on and off every 3 seconds.

reillyeon, 2017-06-27:
This library is not compatible with the Arduino UNO because the application processor on that board does not have control over the USB port. I do not have an Arduino 101 board but looking at the SDK there is likely some work to get this library to work there because the system architecture is different.

closed 2017-06-27 by reillyeon

c0debreaker, 2017-07-02:
Which arduino hardware should I buy that works well with WebUSB?

reillyeon, 2017-08-31:
Boards based on a single ATmega32U4 chip should work. This includes the Arduino Leonardo and Pro Micro boards and their clones such as the Adafruit Feather 32u4 series.

varunkumar, 2017-11-09:
Is this board https://www.arduino.cc/en/Main/ArduinoBoardMegaADK?from=Main.ArduinoBoardADK supported?

jpliew, 2017-11-09:
@varunkumar the MEGA board has a ATmega8U2 chip bridging the USB. This flash and ram in this chip is probably too low to run this library.