webusb/arduino #35: Rassberry PI driver

talbenbasat, 2017-09-08:
Hi,
First thanks for this amazing project!

Is there any plan for developing a driver for rassberry pi?

reillyeon, 2017-09-08:
The Raspberry Pi Zero has a USB controller that supports gadget mode. The Linux kernel has built in support for presenting itself as a number of standard USB class devices including storage, serial and networking. For WebUSB you would want to present a vendor-specific interface by writing a kernel or user-space gadget driver. Exactly what kind of interface you want to present to the USB host depends on what kind of application you are interested in building.

monkey-jsun, 2017-11-04:
@reillyeon is there guide to help create user space gadget driver with vendor specific interface for raspberry pi zero?  In my case I just need two simple bulk-in/bulk-out endpoints.  Really appreciate it.

reillyeon, 2017-11-06:
Years ago I wrote some Python code to make creating userspace gadget drivers easier as part of a Chrome testing project. I don't have a Raspberry Pi Zero to test this with but in theory it should work. The caveat is that in my experience USB gadget support is rather finicky and especially since this has not been tested with the Pi's specific OTG controller there are likely things that need to be tweaked.

https://chromium.googlesource.com/chromium/src/+/master/tools/usb_gadget

It would be really cool if someone developed a library for Raspberry Pi (and other Linux devices with OTG support) that allowed passing of JSON messages over WebUSB similar to a WebSocket.

monkey-jsun, 2017-11-15:
Hi, Reillyeon,

I finally get some time to go through your code.  It is beautifully written, although some parts are over my head :).  Many thanks.

I have a few high-level questions at this point.  Would appreciate your feedback.

- In order to support raspberry pi, it seems I need to replace linux_gadgetfs.py with newer configfs style access, right?
- how is this code intended to be used?  I'm confused by "server" waiting on some interface for HTTP connections.  I was expecting it used as a library by some test programs talking to USB host.  No?
- My goal is to have a composite gadget with 1 ACM interface and 1 generic functionfs interface (1 bulk in endpoint and 1 bulk out endpoint).  How will this be composed?
- In order for the generic functionfs interface to be recognized by windows and use winusb driver, I figure we need to set some kind MS OS descriptor.  There are several related functions in the framework.  Can you highlight steps that need to be done?
- It seems currently we are using a separate thread to handle ep read or write.  Is this still necessary with configfs?

 Thanks.  Again, really great code.  I hope I can understand enough and am able to contribute.

reillyeon, 2017-12-09:
Yes, I think you'll need to update the code to use the more modern configfs interface so that it can coexist with the kernel-based ACM interface. I haven't looked into it so I'm not sure if the multiprocess model (yes, multiprocess, not multithreaded) is still necessary. Hopefully the newer API is also easier to use.

There is code there that sets up the MS OS descriptors but for the 2.0 version I also had to use a kernel patch that changed bcdUSB to 0x0210 so that Windows would read the capability descriptor. The compatibility ID descriptor *might* work if you use the 1.0 version descriptors but I haven't tested that in a while.

The server stuff has to do with the fact that this was designed for testing. The server listens for commands over HTTP and then configures the gadgetfs interface as necessary for the test.