webusb/arduino #36: Strange behavior in demo app

eric-ibarra, 2017-09-26:
Let me start off by saying, that I was able to get everything working correctly with the arduino example with the help of others that posted what they found regarding windows descriptors. I've created another USB device with the appropriate descriptors to be recognized as a webusb device, however, when I try to run the demo javascript application, I get the following error message:

"Connection error: NotFoundError: The interface number provided is not supported by the device in its current configuration."

This happens in the selectAlternateInterface function AFTER the claimInterface function completes without any error messages. Based on what I've found online, it seems like this should never happen since the same check that is performed to find the interface index is the same for both. Am I missing something that would cause this behavior?

reillyeon, 2017-09-27:
The way the promises are chained in the demo app it is impossible to tell which method is triggering the error. I assume you have modified the code in some way to distinguish. It is easy to get this wrong and end up catching errors you don't expect. My guess is that on your device the interface number is different. Can you confirm by looking at the `USBDevice` object that it has an interface number 2?

eric-ibarra, 2017-09-28:
It was actually in the controlTransferOut function. I had changed the interface number in all the other places except for the one inside the control transfer. Thanks for the great example project!

closed 2017-09-28 by eric-ibarra