webusb/arduino #38: Compatibility of Arduino Mega ADK

varunkumar, 2017-11-09:
Is this library compatible with Arduino Mega ADK board? https://www.arduino.cc/en/Main/ArduinoBoardMegaADK?from=Main.ArduinoBoardADK

jpliew, 2017-11-09:
@varunkumar the MEGA board has a ATmega8U2 chip bridging the USB. This flash and ram in this chip is probably too low to run this library.

xander777, 2017-11-10:
I tried to use my chinees copy of MEGA 2560 with console demo app and it didn't work.
maybe someone knows how to fix it or what min parameters of board should be for work with WebUSB?

reillyeon, 2017-11-10:
The UNO and MEGA are built differently from the boards which are supported. Instead of the USB port being directly connected to the microcontroller on which your sketch (and the WebUSB library) runs there is an additional chip which acts as a USB to serial converter and connects to one of the UART pins on the main microcontroller. This means that in order to get WebUSB support on the device you need to reprogram this chip as well and this is currently unsupported.

varunkumar, 2017-11-13:
Thanks for the explanation

closed 2017-11-13 by varunkumar