webusb/arduino #41: Landing-page notification does not appear.

tylercrumpton, 2017-12-07:
After loading either Arduino example to my board, I do not get a notification message with the link to the landing-page. The rest of WebUSB seems to work, because I can use the console example to send messages back and forth to the board.

If I add a new "fake" device in Chrome, via chrome://usb-internals/, the landing-page notification is displayed properly.

**Board:** Adafruit Feather M0
**OS:** Mac OS X 10.13.1
**Arduino version:** 1.8.5
**Output from chrome://device-log/ :** 

    USB device added: vendor=9114 "Adafruit", product=32779 "Feather M0", serial="WUART", guid=193354a7-57da-49bd-b887-6ebc6054c58e

reillyeon, 2017-12-07:
The instructions in README.md only update the USB version for AVR boards. Without the board reporting a USB version of 0x0210 Chrome will not attempt to read the landing page descriptor from the device because it assumes it is not supported. The M0 is an SAMD board.

tylercrumpton, 2017-12-08:
Ah! Thanks, that makes sense. I'll look into trying to update the USB version for the SAMD.

tylercrumpton, 2018-01-30:
@basevers I have not made progress, no. I'll give that change a shot and see how it works on the Mac this afternoon.

tylercrumpton, 2018-02-02:
@basevers So it looks like I may have already had that change in my ``USBCore.h`` file, or it's already in there for me. It's also possible I misunderstood your change, so here is what my line looks like:

```c
#define D_DEVICE(_class,_subClass,_proto,_packetSize0,_vid,_pid,_version,_im,_ip,_is,_configs) \
	{ 18, 1, USB_VERSION, _class,_subClass,_proto,_packetSize0,_vid,_pid,_version,_im,_ip,_is,_configs }
```

I believe that is only for the AVR cores though, so I don't think that affects the Cortex M0-based boards.

Sherek690, 2019-07-30:
Any solution to change the USB_VERSION to 0x210 on SAMD boards ?

flety-emmanuel, 2019-11-15:
Hello, After a solution as well, tried changing the USB version with no success, CDC doesn't mount. Tried on a metro M0 express for a project at work, no dice. Anything to alter in the core ?

gpelizzo, 2020-11-26:
Hello, I've also tried with an Arduino 33 IOT, changing the USB version as mention, but no chance to run WebUsb.  Does someone already succeeded ?