webusb/arduino #42: Not getting webusb notification in Chrome

sandeepks1, 2017-12-20:
I am very new to web USB,To experiment for how exactly webusb works

I bought one Arduino board. Folowed the steps in https://github.com/webusb/arduino Loaded up sketch.ino and program it to my device. 
Here my problem is still am not getting below notification 
[image]

reillyeon, 2017-12-21:
You have not specified what operating system you are using but since the screenshot is from Chrome OS I am going to assume that is where you are. The issue is likely that the device is still not reporting a bcdUSB (USB protocol version) of 0x0210 (2.1). Please double-check that you have modified the headers as instructed.

If you are on Windows then this notification has been disabled due to instabilities in Chrome's code on Windows. Work to fix this is ongoing.

closed 2017-12-21 by reillyeon

sandeepks1, 2017-12-22:
I am using windows 10  OS,How to enable this notification in chrome (Version 63.0.3239)

reillyeon, 2017-12-22:
You cannot currently enable the notification on Windows. It will be re-enabled when the issues it triggers are fixed. The current status is described in [README.md](https://github.com/webusb/arduino/blob/gh-pages/README.md).