webusb/arduino #43: Add DeviceInterfaceGUIDs registry key for Windows

reillyeon, 2017-12-21:
In order for an application to find a USB device on Windows it must have a device interface GUID. For the root USB device this will be GUID_DEVINTERFACE_USB_DEVICE however for functions of a composite device there is no standard GUID. For devices using WinUSB Microsoft recommends adding a descriptor that tells Windows what GUID to use. This value ends up in the DeviceInterfaceGUIDs registry key when the device is enumerated.

gusarov, 2018-03-31:
Looks like this is already implemented?

gusarov, 2018-04-01:
Actually, I still have an issue that sounds exactly like a subject
[image]
I would be glad if you'll help me to fix this or point me to the right direction

jpliew, 2018-04-01:
@gusarov  use regedit and search for 975F44D9-0D08-43FD-8B3E-127CA8AFFF9D first. If nothing then you should retry to download a fresh copy of Arduino and WebUSB to try again. If you find this GUID, you may need to try delete all the references to 975F44D9-0D08-43FD-8B3E-127CA8AFFF9D in the registry first. This appears in a few places, use regedit and search for the above GUID.

Ideally if you can, try on another computer that you have never plug in the board before and test if it is the board problem or windows problem.

Also, what board are you using?

gusarov, 2018-04-02:
Pro Micro (clone)
Thanks, I just noticed that Arduino forgot my CPU setting and reset id back to 3.3v, while mine is 5v :\
Now the next problem - I still can see the COM device when connecting ProMicro with console sketch.., As far as I understand, there should be no more COM port.
PS: I'm flushing via SPI, via second Arduino ISP to avoid messing with USB.

jpliew, 2018-04-02:
@gusarov you should see two devices appear on DEVICE MANAGER. See image below

[image]

gusarov, 2018-04-02:
Thanks, looks like I've messed files too much, so I've started from the scratch. Redownloaded Arduino IDE zip, removed %AppDataLocal%\Arduino15
Just configured package_sparkfun_index.json, downloaded "Spark Fun AVR Boards" via boards manager.
Then I've added "-DUSB_VERSION=0x210" to promicro.build.extra_flags
Uploaded via Arduion as ISP...
And now I have similar picture, but except it is SparkFun Pro Micro instead of 
[image]

gusarov, 2018-04-02:
Similar with Leonardo bootloader:
[image]

gusarov, 2018-04-02:
Finally, I got it... hardware\boards.txt have different vid/pid for the microwebusb, and looks like this is doing the trick... Now I have it in chrome, but Access Denied (looks like I've seen someone had such problem already)

jpliew, 2018-04-02:
@gusarov for access denied, check this out #44

reillyeon, 2018-04-02:
@gusarov Thanks for pointing out that the patch to add the registry key descriptor had been landed. Closing this issue.

closed 2018-04-02 by reillyeon