webusb/arduino #44: Arduino Leonardo Access denied

vinckobb, 2018-01-06:
Hi,

I've tried webusb demo with my Arduino Leonardo but once I try to connect to it I get only Access Denied error.
 
1. Installed Arduino 1.8.5
2. Copied boards definition, library
3. Loaded console sketch and uploaded to Arduino Leonardo
4. Opened up https://webusb.github.io/arduino/demos/console/

And once tried to connect to Leonardo I only get access denied error. Do I have to do anything else to make it run?

OS: Windows 10

basevers, 2018-01-25:
Did you change USB_VER from 0x200 => 0x0210 ?

vinckobb, 2018-01-26:
Thanks for your answer,

I though that information for arduino is already stored in hardware/webusb/avr/boards.txt

I've already tried to change USB_VER from 0x200 => 0x210, still with same results. But I also found error/warning message in Arduino while uploading sample sketch using Arduino Leonardo webusb board.

`Bootloader file specified but missing: {path_to_arduino}\hardware\webusb\avr\bootloaders\caterina\Caterina2-Leonardo-WebUSB.hex`

Do you think this might be the problem? How can I manage it?

basevers, 2018-01-27:
I ordered a Leonardo board, should arrive on Monday. Ill let you know my findings by the middle of next week.

basevers, 2018-01-30:
@vinckobb 
Yesterday, I received an Leonardo board, the example code compiled without warnings. Not sure why you had an issue with the bootloader. You could try to reinstall the IDE.

==

For me, the code still didn't do what I expected it to do.
I expected to independent features:
A) After plugging in the USB device with WebUSB implemented, a Chrome gesture should pop-up to redirect to a webpage
B) Using a web-page (RGB demo), allowing to change the DAC values.

After programming the Leonardo board following the directions (to the best of my knowledge :-) )

1) Chrome 63.0 (Windows 10), didn't asked for the gesture to redirect to the website

chrome://device-log/ showed this:
`
USBDebug[2018/01/30 06:34:29.096255] usb_service_impl.cc:79 Driver for \\?\usb#vid_2341&pid_8036&mi_00#8&19617b49&0&0000#{86e0d1e0-8089-11d0-9ce4-08003e301f73} is usbser.
USBDebug[2018/01/30 06:34:29.090644] usb_service_impl.cc:79 Driver for \\?\usb#vid_2341&pid_8036#wuart#{a5dcbf10-6530-11d2-901f-00c04fb951ed} is usbccgp. 
`

Looking at the captured data using Wireshark (indicating USB 0x0210)
`
DEVICE DESCRIPTOR
    bLength: 18
    bDescriptorType: 0x01 (DEVICE)
    bcdUSB: 0x0210
    bDeviceClass: Miscellaneous (0xef)
    bDeviceSubClass: 2
    bDeviceProtocol: 1 (Interface Association Descriptor)
    bMaxPacketSize0: 64
    idVendor: Arduino SA (0x2341)
    idProduct: Leonardo (CDC ACM, HID) (0x8036)
    bcdDevice: 0x0100
    iManufacturer: 1
    iProduct: 2
    iSerialNumber: 3
    bNumConfigurations: 1
`

2) "No Devices Find"
When at the website:   https://webusb.github.io/arduino/demos/rgb/
Clicking the `connect` button indicated `no devices found`
After fiddling around a little bit, I noticed that `setup` never completed.

_void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, HIGH);
  while (!Serial) {
    ;
  }
  **digitalWrite(LED_BUILTIN, LOW);   <<<<< never got here**
  Serial.begin(9600);
  Serial.write("Sketch begins.\r\n");
  Serial.flush();
  index = 0;
}_

Im not the expert here, but it seems there is an issue with WebUSB and pluggableUSB
I think when this is resolved, it should be working.

==
Just an FYI.
Im able to communicate through a webpage (running a java script) with a usb peripheral. This makes me to believe that the combination of Chrome and Windows 10, i'm using, is in general working.

jpliew, 2018-01-31:
@vinckobb  you can ignore the error

`Bootloader file specified but missing: {path_to_arduino}\hardware\webusb\avr\bootloaders\caterina\Caterina2-Leonardo-WebUSB.hex`

WebUSB is still using the original Caterina bootloader.

vinckobb, 2018-02-01:
I've tried same procedure on another clean computer. Installed Arduino and then added webusb library with sketches. Unfortunately still same results.

@basevers did you also updated USBCore.h file for usb 2.1?

I noticed one thing. When I plugged arduino, chrome didn't find any device. I had to enable _"Enable new USB backend"_ in **chrome://flags**

After that it was recognized and I finally saw it through **chrome://device-log** as
`[19:06:51] usb_service_win.cc:386 USB device added: path=\\?\usb#vid_2341&pid_8036#wuart#{a5dcbf10-6530-11d2-901f-00c04fb951ed} vendor=9025 "Arduino LLC", product=32822 "Arduino Leonardo WebUSB", serial="WUART", driver="usbccgp", guid=94a9678a-4a42-44cd-8a0e-241f8177ea96` 

This doesn't help and I still get _"Access denied"_ error when I try to connect to it.

basevers, 2018-02-01:
@vinckobb 

I learned that the "prefered" method for updating the USB_VERSION is through the board.txt file.
As done here: https://github.com/webusb/arduino/blob/d845253b5c63385b4648ee3ec7eb4be9bbe0f8d1/hardware/webusb/avr/boards.txt#L57

You can verify with WireShark if the USB_VERSION number is the proper one.

@vinckobb 
Do you know if you get passed _setup()_ ??
In my environment, the code is stuck in the _while (!Serial) {;}_ loop.
An easy way to verify is using the onboard LED's, using the following lines:
_pinMode(LED_BUILTIN, OUTPUT);
digitalWrite(LED_BUILTIN, HIGH);
digitalWrite(LED_BUILTIN, LOW);_

I'm curious what you found out.

Thanks a lot.

reillyeon, 2018-02-01:
Make sure you're using the latest version of the library. I recently merged #28 which adds an additional descriptor which sets a registry key giving the WebUSB function a DeviceInterfaceGUID which is necessary for Chrome to find the interface through the Windows API.

As mentioned in README.md the notification is disabled currently on Windows due to instabilities in some Windows USB host drivers.

Enabling Chrome's new USB backend on Windows will improve enumeration reliability but support for opening and making requests to devices is still incomplete so I don't recommend it at this time.

basevers, 2018-02-01:
@reillyeon 

Thank you for the update, however for me it did not fix the issue I'm facing.

@reillyeon 
Is it normal that the code sits in the setup while loop? is it waiting for an external 'something" before it exit the while loop?

In my code, I activate the LED before the while loop and deactivate the LED when it exits the while loop, but in my case, the LED is always on (meaning its still inside the while loop).

Thanks, Bas

jpliew, 2018-02-02:
@basevers how about I retrieve my full flash in HEX and email to you so that you can use ISP programmer to flash into your Leonardo. If still not working, we can start look at the USB driver that was installed.

basevers, 2018-02-02:
@jpliew 

That would be awesome....   bas(dot)evers(at)evers-online(dot)net 

Thanks so much.

vinckobb, 2018-02-02:
@reillyeon I've tried using newest version of WebUSB library but it's still same as before. Unless I enable _"Enable new USB backend"_ chrome doesn't even show device. Otherwise I only get _"Access denied"_ error.

@basevers Here is result from Wireshark. Looks like board.txt file works for me.
```
bLength: 18
bDescriptorType: 0x01 (DEVICE)
bcdUSB: **0x0210**
bDeviceClass: Miscellaneous (0xef)
bDeviceSubClass: 2
bDeviceProtocol: 1 (Interface Association Descriptor)
bMaxPacketSize0: 64
idVendor: Arduino SA (0x2341)
idProduct: Leonardo (CDC ACM, HID) (0x8036)
bcdDevice: 0x0100
iManufacturer: 1
iProduct: 2
iSerialNumber: 3
bNumConfigurations: 1
```

@basevers I'll try to find if I get over setup() this weekend

jpliew, 2018-02-02:
@basevers I put the file on gist easier that way other can also get the file directly, this image is the full flash image from Arduino Leonardo with WebUSB RGB Demo included.
https://gist.github.com/jpliew/3e15229a46d19a2a3f40c431cc90e554

basevers, 2018-02-02:
@jpliew 

Cool, I'll try it this weekend.

basevers, 2018-02-09:
@jpliew 

Finally I got a programmer, allowing me to program the image provided by @jpliew 

`
C:\tools\avr>avrdude -c usbtiny -p m32u4 -u -U flash:w:WebUSBLeonardo.hex
avrdude: AVR device initialized and ready to accept instructions
Reading | ################################################## | 100% 0.02s
avrdude: Device signature = 0x1e9587 (probably m32u4)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "WebUSBLeonardo.hex"
avrdude: input file WebUSBLeonardo.hex auto detected as Intel Hex
avrdude: writing flash (32730 bytes):
Writing | ################################################## | 100% 37.20s
avrdude: 32730 bytes of flash written
avrdude: verifying flash memory against WebUSBLeonardo.hex:
avrdude: load data flash data from input file WebUSBLeonardo.hex:
avrdude: input file WebUSBLeonardo.hex auto detected as Intel Hex
avrdude: input file WebUSBLeonardo.hex contains 32730 bytes
avrdude: reading on-chip flash data:
Reading | ################################################## | 100% 25.09s
avrdude: verifying ...
avrdude: 32730 bytes of flash verified
avrdude done.  Thank you.
`

So far so good. But still no URL popped up in Chrome, and when going manual to `https://webusb.github.io/arduino/demos/`
Still **no devices found** :-(

Chrome reports:
`USB Debug [2018/02/09 16:25:23.921626] usb_service_impl.cc:79 Driver for \\?\usb#vid_2341&pid_8036&mi_00#8&19617b49&0&0000#{86e0d1e0-8089-11d0-9ce4-08003e301f73} is usbser.
USB Debug [2018/02/09 16:25:23.915355] usb_service_impl.cc:79 Driver for \\?\usb#vid_2341&pid_8036#wuart#{a5dcbf10-6530-11d2-901f-00c04fb951ed} is usbccgp.`

WireShark shows something interesting
`
STRING DESCRIPTOR
    bLength: 48
    bDescriptorType: 0x03 (STRING)
    bString: Arduino Leonardo WebUSB
`

while my compiled version showed this:
`
STRING DESCRIPTOR
    bLength: 34
    bDescriptorType: 0x03 (STRING)
    bString: Arduino Leonardo
`

This made me look at Windows 10 Device Manager. Under "Universal Serial Bus devices" a device called "Arduino Leonardo WebUSB" shows up. It is using winusb.sys 
I read some posts about winusb.sys  Let me read those again.

Question...

Should the text "Sketch begins." show in the terminal? using the hex file attached to it doesn't.

basevers, 2018-02-09:
I just tried it on a chromebook :-) it works. the URL notification showed up and directed me to the webpage. Also I was able to connect :-)

This made to to believe its related to the windows 10 driver

basevers, 2018-02-10:
Another update, using the tool [zadig](http://zadig.akeo.ie/) and selecting libusb-win32 resulted Chrome to ask permission to connect. However, claiming the still port failed :-(

jpliew, 2018-02-10:
Hi @basevers the reason why I gave you the HEX file was because I knew your compile environment is not correct. It is hard to help you fix the compile environment without seeing what is inside your system. I would suggest that you only use ARDUINO IDE and not third party IDE. Follow the instruction on this repo again. Try on a new Windows 10 and use another computer to test.

***NOTE: You kept saying no no URL pop up notification in Windows, this is a known issue, meaning no URL pop up notification is the CORRECT behavior for Windows.***

***NOTE 2: DO NOT use Zadig to install the driver for you!!! It will mess up your drivers. Windows 10 by default will be able to detect WebUSB device as WinUSB device***

Now let's look at your Windows driver again, with the HEX I provided, you should see two devices appear on `DEVICE MANAGER`. See image below

[image]

On Zadig (WARNING, do not use Zadig to install driver), you should see two interfaces, one is 

Arduino Leonardo WebUSB (Interface 0), this is the serial com port that you can also see from `DEVICE MANAGER`, and it should have `usbser` driver

[image]

The second interface is Arduino Leonardo WebUSB (Interface 2), this is the WEBUSB and you can also see the same from `DEVICE MANAGER` as `Arduino Leonardo WebUSB`. And the driver for this device should be `WINUSB`

[image]

Check if you got the same as mine, if not, you should try on another computer with Windows 10. Hope this helps.

jpliew, 2018-02-10:
@basevers forgot to reply that the "Sketch begins."  using the following command

`#define Serial WebUSBSerial`
`Serial.write("Sketch begins.\r\n");`

In the RGB demo will output "Sketch begins." to WebUSB host, not the normal Arduino terminal that you use. If you run this RGB demo sketch with the `Console` demo, you can see the text being printed on Chrome's demo console.

basevers, 2018-02-14:
@jpliew 

Thanks, now all is working. It was painful to undo Zadig because I made so many attempts.

Just in case others run into similar issues caused by Zadig, the trick is to keep uninstalling the driver until all the changes are undone.

vinckobb, 2018-02-19:
Hi guys,

I've made progress and I'm also more confused :) Had to reinstall windows on my main computer and today I gave Arduino one more chance. It's working out of the box without any problem. I've tried both samples and they are working properly.

I want to find out what was the problem and why it doesn't work on my windows tablet but I think you can close this issue.

Thanks all

closed 2018-02-22 by vinckobb

TimPietrusky, 2018-06-22:
@basevers What was the problem after all? My situation is:

* Fresh Windows 10
* Chrome 67 with "Enable new USB backend" enabled (otherwise the device will not even show up)
* Latest version of the WebUSB library from this repository

But I still get the "DomException" when I try to connect to the device using WebUSB.

reillyeon, 2018-06-22:
That the device does not enumerate unless you have the "enable new USB backend" flag enabled means that there is likely something wrong with the device descriptors. The new USB backend does not support composite devices yet so it will not work for Arduinos since they have more than once interface (serial and WebUSB). Can you look in chrome://device-log for more detailed errors?

TimPietrusky, 2018-06-27:
@reillyeon "enable new USB backend" *enabled*:

```
[00:02:38] USB device added: path=\\?\usb#vid_2341&pid_8036#wuart#{a5dcbf10-6530-11d2-901f-00c04fb951ed} vendor=9025 "Arduino LLC", product=32822 "Arduino Leonardo", serial="WUART", driver="usbccgp", guid=4417432f-f367-4aa0-82ba-02044ae1e76e

USBError[00:02:38] Failed to open device to read WebUSB descriptors.
```

"enable new USB backend" *disabled*: No log entries :/

jpliew, 2018-06-28:
Hi @TimPietrusky it will be easier to debug if you report what you see on Device Manager and Zadig. Please see my comment here https://github.com/webusb/arduino/issues/44#issuecomment-364642218

TimPietrusky, 2018-06-28:
@jpliew in Device Manager:

[image]

In Zadig (which I never used before), I choose `Options > List all Devices`, in that list I have two entries for the Arduino:

[image]

[image]

TimPietrusky, 2018-06-28:
I found the solution to my problem: https://github.com/webusb/arduino/issues/4#issuecomment-275732790

> The operating system creates a registry entry, named osvc, under this registry key that indicates whether the device supports Microsoft OS Descriptors. If the device does not provide a valid response the first time that the operating system queries it for a Microsoft OS String Descriptor, the operating system will make no further requests for that descriptor.

My problem was, that I used an older version of the WebUSB library from this repo on Mac and thought that connecting this "prepared" Arduino to Windows would just work. Only afterwards I saw that there is an update of the library and I uploaded that to my Arduino AFTER I connected it to Windows for the first time. And as the comment implies: The device has to provide a valid response the FIRST time it is added. And if that is not the case, it will just not work. 

## Solution

* Go into Device Manager
* Find your "Arduino Leonardo" located at "Universial Serial Bus devices"
* Right click on it
* Choose: Uninstall device
* Disconnect the device from your computer
* Connect it again to your computer

Now it can be accessed in the browser with WebUSB. 

## Further investigation

Does this also mean that using Windows 10 as the development platform means, that when someone is flashing the software onto the Arduino, that they also have to remove the Arduino from the Device Manager? Because otherwise the same thing as above applies or not?

reillyeon, 2018-06-28:
Windows includes the device serial number in the registry entries it creates. The WebUSB library changes the serial number from the Arduino's default (UART -> WUART) so this issue only cropped up because your system had seen a device with the older version of the library.

shepherd007, 2018-12-23:
Hello,
I came across some issues when using Chromium and WebUSB on Windows 10. When I'm trying to run WebUSB demo (by connecting Arduino Leonardo) on Chrome 71.0.3578.98 it works fine; however when running demo on the Chromium 73.0.3649.0 I always get "SecurityError: Access denied". I made a lot of experiments with the code on the board recently - I am wondering if it is Chromium issue or just something is already messed in Windows registry that prevents Chromium from accessing device.

Anton-Juul-Naber, 2019-03-26:
THANK YOU @TimPietrusky . I've spend hours trying to find out why my arduino didn't connect. "Reinstalling" it fixed the issue. I didn't use an older bootloader, but i made a lot of other mistakes the first time i flashed it, and i suspect something got saved.

@reillyeon I would seriously suggest that a note is added to the readme, just noting that removing the device in device manager could refresh old information. It seems to me like it could help a lot of people.

puttley, 2019-05-30:
HELLO - I am having similar issues after updating my PC to the latest version of Windows 10 and Chrome (version 74). I set the proper flags in Chrome, and am starting chrome with command line webUSB enable flag set on. Chrome sees the device as paired, but when I try to connect I get "Security Error: Access Denied. Nothing has changed with firmware (using Arduino Leonardo with webUSB library), or the web software. We are in prototype development of a product that is a perfect fit for webUSB, but I am now skeptical about moving forward. Any additional help would be much appreciated. 

Here is what Chrome Device Log is reporting:

USBUser[09:04:37] USB device added: path=\\?\usb#vid_2341&pid_8036#wuart#{a5dcbf10-6530-11d2-901f-00c04fb951ed} vendor=9025 "Arduino LLC", product=32822 "Arduino Leonardo", serial="WUART", driver="usbccgp", guid=9015a91a-ea68-4652-906b-4257a579d0ab

USBError[09:04:37] Failed to open device to read WebUSB descriptors.**

reillyeon, 2019-05-30:
@puttley, please file an issue in the Chromium bug tracker (crbug.com) with details about your device such as interface classes (it looks like your device is composite) as well as exactly which flags you have set (you shouldn't need any) and exactly which version of Chrome and Windows you are running.

puttley, 2019-05-30:
Thank you for your prompt reply. Since posting, I was able to get it to work. I had the "Enable new USB back end" flag set to enabled and also using the command line flag  "--enable-features=WebUSB". With both of these set, my device was being seen as paired but would not connect. I disabled the "Enable new USB back end" flag and now my device connects just fine. I still have to use the command line WebUSB enable flag when starting chrome - so turns out having both enabled caused the issue.

Thank you.

reillyeon, 2019-05-30:
@puttley the WebUSB feature flag is [enabled by default](https://cs.chromium.org/chromium/src/content/public/common/content_features.cc) so you shouldn't need that either. If the feature is still disabled when you don't add the command line flag please file an issue on crbug.com so we can investigate further.

puttley, 2019-05-30:
Thanks again. In my case, I had to add the command line to enable WebUSB in order to get my device to work after updating to the latest version of Chrome (Version 74.0.3729.169 (Official Build) (64-bit)). Interestingly enough, once I launched Chrome with the command line enabled, I can now remove it and my device still works. Seems as though I had to launch Chrome just once with the WebUSB command line set to enable. There is online information about Google disabling webUSB in latest version of Chrome. Maybe it has now been re-enabled by default?

**Google disables WebUSB in Chrome amidst phishing concerns**

Phishing is a major concern if you live a lot of your life on the internet. There are plenty of ways to protect against phishing attacks with software, but one of the best methods is hardware USB keys. An authentication device can protect you even if the attacker has your username and password. At least that’s what they’ll tell you. A pair of researchers proved that these devices are not invincible. A feature in Chrome called WebUSB made it possible to bypass the protections.

WebUSB is a feature that allows websites to directly connect to USB devices; it was added in Chrome 61. Attackers can use the feature with an accompanying website to convince someone to type in their username and password and send it directly to the authentication device to unlock the account. Obviously, which Chrome being the most popular browser on the planet, this is a pretty serious vulnerability.

When asked about it, Google’s security product manager said they are aware of the situation. They consider this type of attack to be an edge case, but they are working to fix the problem. For the time being, Google has disabled the WebUSB feature entirely. This was discovered in Chromium just yesterday. Users can apply a command line flag if they really want to re-enable the feature. For now, the problem has been solved by removing the moving parts.

melwong, 2023-11-15:
> I found the solution to my problem: [#4 (comment)](https://github.com/webusb/arduino/issues/4#issuecomment-275732790)
> 
> > The operating system creates a registry entry, named osvc, under this registry key that indicates whether the device supports Microsoft OS Descriptors. If the device does not provide a valid response the first time that the operating system queries it for a Microsoft OS String Descriptor, the operating system will make no further requests for that descriptor.
> 
> My problem was, that I used an older version of the WebUSB library from this repo on Mac and thought that connecting this "prepared" Arduino to Windows would just work. Only afterwards I saw that there is an update of the library and I uploaded that to my Arduino AFTER I connected it to Windows for the first time. And as the comment implies: The device has to provide a valid response the FIRST time it is added. And if that is not the case, it will just not work.
> 
> ## Solution
> * Go into Device Manager
> * Find your "Arduino Leonardo" located at "Universial Serial Bus devices"
> * Right click on it
> * Choose: Uninstall device
> * Disconnect the device from your computer
> * Connect it again to your computer
> 
> Now it can be accessed in the browser with WebUSB.
> 
> ## Further investigation
> Does this also mean that using Windows 10 as the development platform means, that when someone is flashing the software onto the Arduino, that they also have to remove the Arduino from the Device Manager? Because otherwise the same thing as above applies or not?

This worked for me.