webusb/arduino #48: A little doc?

ANSB, 2018-04-04:
I use the Lib and it works fine, despite the lack of coment in the examples which made them not very easy to understand. :)
My question is about the WebUSB WebUSBSerial(1, "url"); call. I notice the url is not mandatory and if you don't add it, Chrome don't display its small Box. But what about the first parameter? Can you explain its meaning and possible values? 
Thanks!

reillyeon, 2018-04-04:
I just added documentation for this parameter last week:

    /*
     * Together |landingPageScheme| and |landingPageUrl| tell the browser
     * what page the user should visit in order to interact with their
     * device. |landingPageScheme| can have any of the following values:
     *
     *  0x00 -> "http://"
     *  0x01 -> "https://"
     *
     * This prefix is combined with |landingPageUrl| to produce the full
     * URL.
     */
    WebUSB(uint8_t landingPageScheme, const char* landingPageUrl);

(From https://github.com/webusb/arduino/blob/gh-pages/library/WebUSB/WebUSB.h#L51)

closed 2018-04-04 by reillyeon

reillyeon, 2018-04-04:
http:// is allowed because according to the specification http://localhost is considered a secure context.