webusb/arduino #57: Chrome randomly unable to read vendor, product or serial description

jpliew, 2018-11-21:
@reillyeon, I am not sure if this is an issue with Chrome, Arduino WebUSB or Windows, if you have any idea where should I start to debug, please share.

Device :  Arduino Leonardo
WebUSB library : Current
Chrome : Version 70.0.3538.110 (Official Build) (64-bit)
Windows : 10 and 8.1 

I noticed at times, chrome://device-log have empty vendor, product or serial description. I did 12 times of plug in and unplug test and Chrome will randomly returns empty for these descriptions. At all time, the device is working properly with the demos (except not showing the device name when Chrome does not show on device-log), Windows device manager show the COM and WEBUSB device detected properly.

```
USBDebug[15:23:06] Driver for \\?\usb#vid_2341&pid_8036&mi_02#6&19617b49&0&0002#{dee824ef-729b-4a0e-9c14-b7117d33a817} is WINUSB.

USBUser[15:23:06] USB device added: vendor=9025 "", product=32822 "Arduino Leonardo WebUSB", serial="", guid=3a0a3e06-b5fe-45f1-abca-7110b7f8c579

USBEvent[15:23:06] Failed to submit transfer: Entity not found

USBDebug[15:23:06] Driver for \\?\usb#vid_2341&pid_8036&mi_02#6&19617b49&0&0002#{975f44d9-0d08-43fd-8b3e-127ca8afff9d} is WINUSB.

USBDebug[15:23:06] Driver for \\?\usb#vid_2341&pid_8036&mi_00#6&19617b49&0&0000#{86e0d1e0-8089-11d0-9ce4-08003e301f73} is usbser.

USBDebug[15:23:06] Driver for \\?\usb#vid_2341&pid_8036#wuart#{a5dcbf10-6530-11d2-901f-00c04fb951ed} is usbccgp.

USBUser[15:23:02] USB device removed: guid=6b9e4085-ed46-4630-958c-b55329aa3cc5

USBDebug[15:22:55] Driver for \\?\usb#vid_2341&pid_8036&mi_02#6&19617b49&0&0002#{dee824ef-729b-4a0e-9c14-b7117d33a817} is WINUSB.

USBUser[15:22:55] USB device added: vendor=9025 "Arduino LLC", product=32822 "Arduino Leonardo WebUSB", serial="WUART", guid=6b9e4085-ed46-4630-958c-b55329aa3cc5

USBDebug[15:22:55] Driver for \\?\usb#vid_2341&pid_8036&mi_02#6&19617b49&0&0002#{975f44d9-0d08-43fd-8b3e-127ca8afff9d} is WINUSB.

USBDebug[15:22:55] Driver for \\?\usb#vid_2341&pid_8036&mi_00#6&19617b49&0&0000#{86e0d1e0-8089-11d0-9ce4-08003e301f73} is usbser.

USBDebug[15:22:55] Driver for \\?\usb#vid_2341&pid_8036#wuart#{a5dcbf10-6530-11d2-901f-00c04fb951ed} is usbccgp.

USBUser[15:22:52] USB device removed: guid=4d85f157-9e54-41ca-a62f-18eb7ee15f33

USBDebug[15:22:26] Driver for \\?\usb#vid_2341&pid_8036&mi_02#6&19617b49&0&0002#{dee824ef-729b-4a0e-9c14-b7117d33a817} is WINUSB.

USBUser[15:22:26] USB device added: vendor=9025 "", product=32822 "Arduino Leonardo WebUSB", serial="WUART", guid=4d85f157-9e54-41ca-a62f-18eb7ee15f33

USBEvent[15:22:26] Failed to submit transfer: Entity not found

USBDebug[15:22:26] Driver for \\?\usb#vid_2341&pid_8036&mi_02#6&19617b49&0&0002#{975f44d9-0d08-43fd-8b3e-127ca8afff9d} is WINUSB.

USBDebug[15:22:26] Driver for \\?\usb#vid_2341&pid_8036&mi_00#6&19617b49&0&0000#{86e0d1e0-8089-11d0-9ce4-08003e301f73} is usbser.

USBDebug[15:22:26] Driver for \\?\usb#vid_2341&pid_8036#wuart#{a5dcbf10-6530-11d2-901f-00c04fb951ed} is usbccgp.

USBUser[15:22:23] USB device removed: guid=233d71b3-6348-43b3-a2da-a4d97337d744

USBDebug[15:22:15] Driver for \\?\usb#vid_2341&pid_8036&mi_02#6&19617b49&0&0002#{dee824ef-729b-4a0e-9c14-b7117d33a817} is WINUSB.

USBUser[15:22:15] USB device added: vendor=9025 "", product=32822 "Arduino Leonardo WebUSB", serial="WUART", guid=233d71b3-6348-43b3-a2da-a4d97337d744

USBEvent[15:22:15] Failed to submit transfer: Entity not found

USBDebug[15:22:15] Driver for \\?\usb#vid_2341&pid_8036&mi_02#6&19617b49&0&0002#{975f44d9-0d08-43fd-8b3e-127ca8afff9d} is WINUSB.

USBDebug[15:22:15] Driver for \\?\usb#vid_2341&pid_8036&mi_00#6&19617b49&0&0000#{86e0d1e0-8089-11d0-9ce4-08003e301f73} is usbser.

USBDebug[15:22:15] Driver for \\?\usb#vid_2341&pid_8036#wuart#{a5dcbf10-6530-11d2-901f-00c04fb951ed} is usbccgp.

USBUser[15:22:11] USB device removed: guid=3b572545-dd80-4bde-b5e3-322a4f920f84

USBDebug[15:22:03] Driver for \\?\usb#vid_2341&pid_8036&mi_02#6&19617b49&0&0002#{dee824ef-729b-4a0e-9c14-b7117d33a817} is WINUSB.

USBUser[15:22:03] USB device added: vendor=9025 "Arduino LLC", product=32822 "Arduino Leonardo WebUSB", serial="WUART", guid=3b572545-dd80-4bde-b5e3-322a4f920f84

USBDebug[15:22:03] Driver for \\?\usb#vid_2341&pid_8036&mi_02#6&19617b49&0&0002#{975f44d9-0d08-43fd-8b3e-127ca8afff9d} is WINUSB.

[... 77 lines cut ...]

USBUser[15:20:43] USB device added: vendor=9025 "", product=32822 "Arduino Leonardo WebUSB", serial="WUART", guid=7b286e51-e40d-4f99-9954-ae6af3f5eda6

USBEvent[15:20:43] Failed to submit transfer: Entity not found

USBDebug[15:20:43] Driver for \\?\usb#vid_2341&pid_8036&mi_02#6&19617b49&0&0002#{975f44d9-0d08-43fd-8b3e-127ca8afff9d} is WINUSB.

USBDebug[15:20:43] Driver for \\?\usb#vid_2341&pid_8036&mi_00#6&19617b49&0&0000#{86e0d1e0-8089-11d0-9ce4-08003e301f73} is usbser.

USBDebug[15:20:43] Driver for \\?\usb#vid_2341&pid_8036#wuart#{a5dcbf10-6530-11d2-901f-00c04fb951ed} is usbccgp.

USBUser[15:20:39] USB device removed: guid=a5df0b83-d501-4ab7-9001-55a270933627

USBDebug[15:20:17] Driver for \\?\usb#vid_2341&pid_8036&mi_02#6&19617b49&0&0002#{dee824ef-729b-4a0e-9c14-b7117d33a817} is WINUSB.

USBUser[15:20:17] USB device added: vendor=9025 "", product=32822 "Arduino Leonardo WebUSB", serial="WUART", guid=a5df0b83-d501-4ab7-9001-55a270933627

USBEvent[15:20:17] Failed to submit transfer: Entity not found

USBDebug[15:20:17] Driver for \\?\usb#vid_2341&pid_8036&mi_02#6&19617b49&0&0002#{975f44d9-0d08-43fd-8b3e-127ca8afff9d} is WINUSB.

USBDebug[15:20:17] Driver for \\?\usb#vid_2341&pid_8036&mi_00#6&19617b49&0&0000#{86e0d1e0-8089-11d0-9ce4-08003e301f73} is usbser.

USBDebug[15:20:17] Driver for \\?\usb#vid_2341&pid_8036#wuart#{a5dcbf10-6530-11d2-901f-00c04fb951ed} is usbccgp.
```

jpliew, 2018-11-21:
Sorry for the long log, below is just the highlights of the issue

```
USBUser[15:23:06] USB device added: vendor=9025 "", product=32822 "Arduino Leonardo WebUSB", serial="", guid=3a0a3e06-b5fe-45f1-abca-7110b7f8c579
USBUser[15:22:55] USB device added: vendor=9025 "Arduino LLC", product=32822 "Arduino Leonardo WebUSB", serial="WUART", guid=6b9e4085-ed46-4630-958c-b55329aa3cc5
USBUser[15:22:26] USB device added: vendor=9025 "", product=32822 "Arduino Leonardo WebUSB", serial="WUART", guid=4d85f157-9e54-41ca-a62f-18eb7ee15f33
USBUser[15:22:15] USB device added: vendor=9025 "", product=32822 "Arduino Leonardo WebUSB", serial="WUART", guid=233d71b3-6348-43b3-a2da-a4d97337d744
USBUser[15:22:03] USB device added: vendor=9025 "Arduino LLC", product=32822 "Arduino Leonardo WebUSB", serial="WUART", guid=3b572545-dd80-4bde-b5e3-322a4f920f84
USBUser[15:21:51] USB device added: vendor=9025 "Arduino LLC", product=32822 "Arduino Leonardo WebUSB", serial="WUART", guid=bb0730ba-0739-4f73-acb2-0c5e3e0b0327
USBUser[15:21:41] USB device added: vendor=9025 "", product=32822 "Arduino Leonardo WebUSB", serial="WUART", guid=e819014a-68ef-432b-8485-f8000f3d04bb
USBUser[15:21:28] USB device added: vendor=9025 "", product=32822 "Arduino Leonardo WebUSB", serial="WUART", guid=b395001c-4d19-4aea-8708-747b061ae866
USBUser[15:21:13] USB device added: vendor=9025 "", product=32822 "Arduino Leonardo WebUSB", serial="", guid=fd61c03e-9c53-40b1-9fb7-068c34a5c3f5
USBUser[15:20:59] USB device added: vendor=9025 "", product=32822 "Arduino Leonardo WebUSB", serial="WUART", guid=ffa86fbb-50cf-4808-9a5d-5d7fb9983bb3
USBUser[15:20:43] USB device added: vendor=9025 "", product=32822 "Arduino Leonardo WebUSB", serial="WUART", guid=7b286e51-e40d-4f99-9954-ae6af3f5eda6
USBUser[15:20:17] USB device added: vendor=9025 "", product=32822 "Arduino Leonardo WebUSB", serial="WUART", guid=a5df0b83-d501-4ab7-9001-55a270933627

```

reillyeon, 2018-11-21:
This is a known issue with Chrome on Windows which will be resolved as part of [issue 637404](https://bugs.chromium.org/p/chromium/issues/detail?id=637404).

closed 2018-11-21 by reillyeon