webusb/arduino #58: Are 4 endpoints really neccessary?

shepherd007, 2018-11-28:
I am trying to understand the demo - from what I can see there are two devices created - CDC (with 2 endpoints) and WebUSB (also with two endpoints). There is a comment saying that:

> only try to send bytes if the high-level CDC connection itself is open (not just the pipe)

Does it mean we need to have twice more endpoints than for native (i.e. C++ Windows) serial comm implementation? Can number of endpoints be limited to minimum (which is 2 in the demo)?

jpliew, 2018-11-28:
@shepherd007 this WebUSB demo only requires two endpoints.

The CDC is actually inherited from Arduino, which is what this WebUSB sketch demo depending on. Due to the Arduino Leonardo's bootloading nature that requires a CDC serial to exist for triggering of the bootloader.

It is possible to only have 2 endpoints if you can port this demo to LUFA or you remove most of the core CDC feature from Arduino.

Hope this answer your question.

Cheers
JP

reillyeon, 2018-11-28:
@jpliew's answer is correct.

closed 2018-11-28 by reillyeon

shepherd007, 2018-11-29:
Thank you; that makes a sense.