webusb/arduino #61: How to make this lib to support ESP 32 and STM32F103c8t6 ?

NguyenBaoXuanTruong, 2019-06-04:
Dear all,

After I ran the RGB demo. It worked so well and made me quite excited. I hope I can run this demo on ESP 32 and STM32F1. Here are some of my questions:

Is it possible to make this library support the ESP 32 and STM32F1? How to do it? Do you have instructions to do that?

Thank you.
Truong Nguyen

AdrianDiazCode, 2019-11-15:
exactly I want this library for esp8266

jpliew, 2020-12-30:
For STM32F1 please refer to https://gitlab.eurecom.fr/manfredi/WebUSB/-/wikis/blue-pill-webusb-dfu

For ESP32 or ESP8266, because these two chips have no native USB port, therefore WebUSB will not work. In my opinion, chip with WIFI should not even look at the WebUSB, it will be easier to have the chip talk to the web app via TCP/IP.

scott-olero, 2021-02-24:
Chekout TINY USB

closed 2021-03-08 by nbxtruong