webusb/arduino #62: USB descriptor information not shown correctly in Chrome/Chromium WebRTC

shepherd007, 2019-06-26:
When I'm trying to run example console demo using Arduino Leonardo I can see that navigator.usb.getDevices() doesn't seem to work correctly. It seems to return valid product name (Arduino Leonardo) but no serial number (USBLyzer tool shows there is a serial number: "WUART") and no manufacturer field. It looks to me there can be some kind of the issue in the Chromium implementation which I'd like to confirm (or something missing in the USB descriptors) related to Windows 10.

The issue can be reproduced on the Windows 10 in another way:
1. connect Arduino Leonardo with WebUSB sketch uploaded
2. start console demo page
3. allow using Arduino Leonardo
4. go to second Chromium tab and navigate to: chrome://usb-internals
5. on the 'Devices' tab you;ll see Arduino Leonardo with empty fields (such as: serial number)

The navigator.usb.getDevices() seem to return all data on the Linux (i.e. CentOS 7)

jpliew, 2019-06-26:
On Windows, intermittently will have missing fields. See this
https://github.com/webusb/arduino/issues/57

If your application is running on Windows, do not make it depending on anything other than the VID and PID.

reillyeon, 2019-06-26:
This is Chromium [issue 637404](https://crbug.com/637404). The problem is that on Windows and macOS Chromium currently uses libusb to communicate with USB devices rather than the native OS API. libusb does not provide a way of reading the string descriptors cached by the operating system and so Chromium must try to read them itself directly from the device. This is not reliable as the device may not be ready for connections so soon after being connected. There is an ongoing project to replace libusb on Windows with code using the native API directly. It has taken longer than I'd like to find the time to complete it.

closed 2019-06-26 by reillyeon

shepherd007, 2019-06-27:
@reillyeon Thanks for clarification. It sounds like an issue in Windows implementation of the libusb library.