webusb/arduino #66: Unable to get it working with Feather M0 Express (SAMD21) despite demo by Adafruit

armsp, 2019-08-12:
I am on Ubuntu 18.04 with Chrome 76. I am completely new to this and have been fiddling with it for a day but I just can't get WebUSB working with my [Feather M0 Express](https://www.adafruit.com/product/3403) that uses SAMD21 or my nRF52840 based Arduino Nano 33 BLE.
In the list of [Compatible Hardware] in WebUSB's Readme (https://github.com/webusb/arduino/#compatible-hardware) `Arduino MKRZero` is mentioned that has a SAMD21, so I assumed that it should work on my board too.
I believe I have followed all the correct instructions in the Getting Started.
Moreover there are couple of videos by Adafruit where they show WebUSB demo on boards that use SAMD21, however there just isn't more details about how they got it working - [WebUSB1](https://www.youtube.com/watch?v=brZ9s8zAJZg), [WebUSB2](https://www.youtube.com/watch?v=5UU8GRoDw8c)

There is a [tinyUSB page](https://github.com/hathach/tinyusb) AND in the Arduino IDE if you click on **Tools** then an option called `USB Stack` comes where you can choose `Arduino` or `TinyUSB`. This has been confusing me. Moreover on TinyUSB's [Supported Boards](https://github.com/hathach/tinyusb#supported-mcus) they have mentioned both SAMD21 and nRF52840. This is incredibly confusing to me. And then there is a [Adafruit TinyUSB Library](https://github.com/adafruit/Adafruit_TinyUSB_Arduino) that doesn't have much documentation and a [separate core for SAMD devices](https://github.com/adafruit/ArduinoCore-samd). But how do I fit this information to get things working.

While trying WebUSB RGB demo from Arduino IDE, I even tried changing the `USB Stack` to `TinyUSB` but that made the code fail to even compile.

Could you please shed some light here?
I am willing to do my research and get it working, but I was hoping you could clear this hodge-podge of information that I have into something discernible.

reillyeon, 2019-08-12:
@sandeepmistry for the SAMD support. I am not familiar with these boards.

This library uses the Arduino USB stack, not TinyUSB. Adafruit's video links to [this example code](https://github.com/hathach/tinyusb/blob/master/examples/device/webusb_serial/src/main.c) for using WebUSB with TinyUSB.

armsp, 2019-08-24:
@sandeepmistry any hint on this? 
So far I have tried -
1. Adding the udev rule as mentioned [here](https://developers.google.com/web/updates/2016/03/access-usb-devices-on-the-web#tips) in a file `50-adafruit_m0_express.rules` -
```SUBSYSTEM=="usb", ATTR{idVendor}=="2341", MODE="0664", GROUP="plugdev"```
2. Added my user to `plugdev`
3. Started Chrome with ```--disable-webusb-security```

Still then my browser doesn't recognize my Feather M0 Express. Even the Adafruit's own examples pointed by @reillyeon don't work. All I see is this -
[image]

sayanee, 2019-08-25:
@armsp What is your `vendorID` and `productID` using [`lsusb` command on Ubuntu](https://wiki.debian.org/HowToIdentifyADevice/USB) when you connect the board to your laptop?

The [RGB demo](https://webusb.github.io/arduino/demos/rgb/) calls [serial.js](https://webusb.github.io/arduino/demos/serial.js) which has a specified list of `vendorID` and `productID` and your board must match, else it will say *No compatible devices found*.

[image]

armsp, 2019-08-25:
My `vendorId` is `0x239A` and productId is `0x801B`. Does `serial.js` check for both `productId` and `vendorId` cause the `vendorId` matches with the last two values - Adafruit ItsyBitsy 32u4 - but the `productId` doesn't match with anything.

armsp, 2019-08-25:
It worked !! You are a rockstar @sayanee

closed 2019-08-25 by armsp