webusb/arduino #69: Notification doesn't appear in Chrome but board connects automatically

armsp, 2019-09-03:
I have a strange issue. Everything is working fine. The board connects, the data transfer happens, the demos work but I **never** get a notification from Chrome.  
I have to manually goto the website and when I do I see my device automatically Connected.

What is even more mind boggling is that when I run WebUSB examples from Adafruit_TinyUSB Library then I **do** get the notifications.
Any clue why it behaves this way?

I am testing it on Ubuntu 18.04.3 LTS with Chrome 76.0.3809.100

armsp, 2019-09-03:
Ok, so I used `chrome://usb-internals` to inspect my device. And in the `USB Version` it shows `2.0.0`.  
But I have already made this change ```#define USB_VERSION 0x210```  
Is it possible that for custom SAMD21 boards, the `USBCore.h` files used are in a location other than ```hardware/arduino/avr/cores/arduino/USBCore.h``` ?  

My arduino device is Adafruit Feather M0 Express

ladyada, 2019-09-03:
if it works in tinyusb, then that's the extent of what adafruit can support with. we dont have any insight into chrome

armsp, 2019-09-03:
I think the issue is deeper than I thought. Some disparities from chrome://usb-internals -

| Arduino WebUSB | Adafruit WebUSB |
| --- | --- |
| USB Version: 2.0.0 | USB Version: 32.1.0 |
| WebUSB Landing Page: <empty>| WebUSB Landing Page: <adafruit's demo page> |

reillyeon, 2019-09-03:
Chrome will only query the device for a landing page if it reports USB 2.1 or greater as that is the version at which the Binary device Object Store descriptor was added to the specification.

armsp, 2019-09-03:
@reillyeon Yeah, but I am not able to understand why the changes in the `USBCore.h` file isn't making it work. Any hints on that ? 
Because the _same hardware_ shows notification when used with Adafruit's library but not with yours despite the changes to `USBCore.h`

jpliew, 2019-09-04:
@armsp it is best for custom `mycustomboard`, the configuration is done at the "boards.txt" file for that board by creating a new board definition.

For example

`mycustomboard.build.extra_flags={build.usb_flags} "-DUSB_VERSION=0x210"`

This flag will be passed to compiler and be used everywhere when a USB_VERSION is required. When uploading sketch, user just need to select the correct `mycustomboard`

Therefore, in order to use ***Adafruit Feather M0 Express*** , I would encourage @ladyada and Adafruit team to add a new board definition for this board and reduce customer frustration editing the `USBCore.h` which I think should not be touch because hardcode `USB_VERSION` in `USBCore.h` will cause other boards failed to operate. 

This issue has been reported here #32

ladyada, 2019-09-04:
does the arduino SAMD definition have it fixed?

armsp, 2019-09-04:
@jpliew I understand. So I thought I'd follow what you say. But the issue is that I can't seem to find any core folders for samd21 or any other architecture. I have installed Arduino Samd Boards from the board manager and Adafruit's boards too but where do they reside locally? Inside `hardware` -> `arduino` the only option I have is `avr`.  
Any idea where exactly I would find the `boards.txt` files for SAMD21 boards, doesn't matter whether they are from Arduino or Adafruit ?

jpliew, 2019-09-04:
@armsp I don't have Adafruit board to test, but I managed to do a dry run and the steps are below.

1. Follow Adafruit's instruction to add board manager url to Arduino IDE
https://adafruit.github.io/arduino-board-index/package_adafruit_index.json

2. Then, Tools, Board Manager, search for Adafruit and install Adafruit SAMD version 1.5.3

3. After install, if you are using Windows, the files will be located at

`C:\Users\<username>\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.3`

If you are on Linux, Arduino IDE, File, Preference, check the bottom part, you should see something like `Arduino15` , this is were the boards files are installed. 

If still can't find it, on Linux just do a `find` command to search for `boards.txt`

***Adafruit Feather M0 Express*** should have the following details

```
# Adafruit Feather M0 Express (SAMD21)
# ------------------------------
adafruit_feather_m0_express.name=Adafruit Feather M0 Express
adafruit_feather_m0_express.vid.0=0x239A
adafruit_feather_m0_express.pid.0=0x801B
adafruit_feather_m0_express.vid.1=0x239A
adafruit_feather_m0_express.pid.1=0x001B
adafruit_feather_m0_express.upload.tool=bossac
adafruit_feather_m0_express.upload.protocol=sam-ba
adafruit_feather_m0_express.upload.maximum_size=262144
adafruit_feather_m0_express.upload.offset=0x2000
adafruit_feather_m0_express.upload.use_1200bps_touch=true
adafruit_feather_m0_express.upload.wait_for_upload_port=true
adafruit_feather_m0_express.upload.native_usb=true
adafruit_feather_m0_express.build.mcu=cortex-m0plus
adafruit_feather_m0_express.build.f_cpu=48000000L
adafruit_feather_m0_express.build.usb_product="Feather M0 Express"
adafruit_feather_m0_express.build.usb_manufacturer="Adafruit"
adafruit_feather_m0_express.build.board=SAMD_FEATHER_M0_EXPRESS
adafruit_feather_m0_express.build.core=arduino
adafruit_feather_m0_express.build.extra_flags=-DARDUINO_SAMD_ZERO -DARDUINO_SAMD_FEATHER_M0 -DARM_MATH_CM0PLUS -DADAFRUIT_FEATHER_M0_EXPRESS -D__SAMD21G18A__ {build.usb_flags}
adafruit_feather_m0_express.build.ldscript=linker_scripts/gcc/flash_with_bootloader.ld
adafruit_feather_m0_express.build.openocdscript=openocd_scripts/feather_m0_express.cfg
adafruit_feather_m0_express.build.variant=feather_m0_express
adafruit_feather_m0_express.build.variant_system_lib=
adafruit_feather_m0_express.build.vid=0x239A
adafruit_feather_m0_express.build.pid=0x801B
adafruit_feather_m0_express.bootloader.tool=openocd
adafruit_feather_m0_express.bootloader.file=featherM0/bootloader-feather_m0-v2.0.0-adafruit.5.bin
adafruit_feather_m0_express.menu.usbstack.arduino=Arduino
adafruit_feather_m0_express.menu.usbstack.tinyusb=TinyUSB
adafruit_feather_m0_express.menu.usbstack.tinyusb.build.flags.usbstack=-DUSE_TINYUSB
adafruit_feather_m0_express.menu.debug.off=Off
adafruit_feather_m0_express.menu.debug.on=On
adafruit_feather_m0_express.menu.debug.on.build.flags.debug=-g
```

I suggest you duplicate these details and create a board call

***Adafruit Feather M0 Express WebUSB***

replacing

`adafruit_feather_m0_express`

with 

`adafruit_feather_m0_express_webusb`

and product name to add WebUSB text there to differentiate the board.

then add the `USB_VERSION` at

`adafruit_feather_m0_express.build.extra_flags=-DARDUINO_SAMD_ZERO -DARDUINO_SAMD_FEATHER_M0 -DARM_MATH_CM0PLUS -DADAFRUIT_FEATHER_M0_EXPRESS -D__SAMD21G18A__ {build.usb_flags}
 "-DUSB_VERSION=0x210"`

Once done, restart Arduino IDE, and you can see your new board showing on Tools, Boards.

Ideally it is preferred to work with @ladyada to add this so that it will be consistent with their product support policy and reduce confusion.

Once this is done, you will have a normal 

***Adafruit Feather M0 Express*** 
work with current Adafruit tutorials and examples.

and ***Adafruit Feather M0 Express WebUSB*** 
work with official WebUSB library examples.

Hope this helps.

Cheers
JP

jpliew, 2019-09-04:
@armsp depending on Adafruit's decision, they might prefer to use their current method

`adafruit_feather_m0_express.menu.usbstack.webusb=xxxx`

no matter which method, still, working together with Adafruit to achieve this is the best, that way Adafruit can provide greater support for customers like you.

armsp, 2019-09-04:
@jpliew thanks a lot for such a detailed explanation and steps. I tried with both the changes but it still doesn't give notifications with Arduino Webusb.
* ```adafruit_feather_m0_express.build.extra_flags=-DARDUINO_SAMD_ZERO -DARDUINO_SAMD_FEATHER_M0 -DARM_MATH_CM0PLUS -DADAFRUIT_FEATHER_M0_EXPRESS -D__SAMD21G18A__ {build.usb_flags} -DUSB_VERSION=0x210```
* ```adafruit_feather_m0_express.build.extra_flags=-DARDUINO_SAMD_ZERO -DARDUINO_SAMD_FEATHER_M0 -DARM_MATH_CM0PLUS -DADAFRUIT_FEATHER_M0_EXPRESS -D__SAMD21G18A__ {build.usb_flags} "-DUSB_VERSION=0x210"```

ladyada, 2019-09-04:
sounds resolved! thaks everyone :)

armsp, 2019-09-04:
@ladyada not really. I really appreciate your time and effort and I understand that you'd want us to use Adafruit TinyUSB Library, but fundamentally it's just a SAMD21, so it should work with Arduino Webusb too, which it does....except the usb version issue that prevents chrome from generating a notification.

jpliew, 2019-09-04:
@ladyada not resolved yet. 

Further checking, Adafruit's SAMD might not be using the original Arduino `USBCore.h` therefore adding `USB_VERSION` in the boards.txt will not work.

@armsp since Adafruit's board is working with TinyUSB, is there any specific reason you want this board to work with the original WebUSB library? 

If you still want to make this work, for learning purposes, you can try hack the Arduino SAMD boards, by downloading from Board Manager. Then try again with my method.

ladyada, 2019-09-04:
for adafruit boards use tinyusb - it works really well :)

ladyada, 2019-09-04:
if you find a solution let us know, a PR will be accepted for sure!

jpliew, 2019-09-04:
@armsp the original Arduino SAMD `USBCore.h` file is at 

`Arduino15\packages\arduino\hardware\samd\1.8.3\cores\arduino\USB`

The USB version is hardcoded to 0x200, you can try change this 

`#define D_DEVICE(_class,_subClass,_proto,_packetSize0,_vid,_pid,_version,_im,_ip,_is,_configs) \
	{ 18, 1, 0x200, _class,_subClass,_proto,_packetSize0,_vid,_pid,_version,_im,_ip,_is,_configs }`

to 

`#define D_DEVICE(_class,_subClass,_proto,_packetSize0,_vid,_pid,_version,_im,_ip,_is,_configs) \
	{ 18, 1, 0x210, _class,_subClass,_proto,_packetSize0,_vid,_pid,_version,_im,_ip,_is,_configs }`

When uploading sketch, please REMEMBER to select ***Arduino SAMD M0*** type of board, NOT Adafruit board.

armsp, 2019-09-04:
I took a slightly different approach than yours - instead of messing around with Arduino's SAMD core I thought since I will choose _Adafruit Feather M0 Express_ anyway from the _Board Manager_ i decided to change the usb files inside adafruit's samd core. Thats located at ```.arduino15\packages\adafruit\hardware\samd\1.5.3\cores\arduino\usb``` . Inside the `USBCore.h` file I made exactly the same change as you suggest before you replied.

> When uploading sketch, please REMEMBER to select Arduino SAMD M0 type of board, NOT Adafruit board.

As I said I want to avoid that. Moreover choosing any Arduino Samd based from board manager before uploading isn't the right approach and it fails anyways except `ArduinoMRKZERO`.
### But I have had mixed success

Now in my ```chrome://usb-internals``` i can see that it detects it as a **USB Version: 2.1.0** device **HOWEVER** the notification still **DID NOT ** come. 
What is crazier is that when I use the `TinyUSB` USB Stack with `Adafruit_TinyUSB` _Webusb_ example, the notification appears immediately **BUT** the **USB Version** has remained **32.1.0**

jpliew, 2019-09-04:
> As I said I want to avoid that. Moreover choosing any Arduino Samd based from board manager before uploading isn't the right approach and it fails anyways except ArduinoMRKZERO.

@armsp why is it not the right approach? When did you say you wanted to avoid that? What fail? upload fail? compile fail? Screenshot?

If you suspect url problem, you can check using

`chrome://usb-internals/`

Click `Devices`, then click `Inspect`. You can see the URL from this screen, please paste your screen shot here.

I can ensure you that this current version of WebUSB is solid and I have many production devices running.

armsp, 2019-09-04:
@jpliew I didn't mean to offend. 
What I meant was that since Adafruit has already taken the pain to make their core and boards it only makes sense to use them as the board pinouts would match incase I want to do something complex later on. And anyways we found where the Adafruit samd core is and the `USBCore.h` file over there.

> If you suspect url problem, you can check using
```chrome://usb-internals/```

I have been giving these information already using ```chrome://usb-internals```

Screenshot of `chrome://usb-internals` when I load arduino webusb example sketch (after `USBCore.h` edit it does show the version as 2.1.0 but **NO notification**) -
[image]

Screenshot of `chrome://usb-internals` when I load Adafruit TinyUSB webusb example sketch and when the notification appears -
[image]

ERRORS when choosing wrong boards
----------------------------------------------------
### Arduino M0
[image]
### Arduino MKRZERO
Uploads but no notification.
USB Version comes as 2.0.0 cause I chose not to change my arduino samd core `USBCore.h` file.    `chrome://usb-internals` screenshot
[image]
### Arduino M0 Pro - Native port
Same error as M0
### Arduino M0 Pro -> Programming Port (as expected)
```
Arduino: 1.8.9 (Linux), Board: "Arduino M0 Pro (Programming Port)"

Build options changed, rebuilding all
Sketch uses 13176 bytes (5%) of program storage space. Maximum is 262144 bytes.
An error occurred while uploading the sketch
Open On-Chip Debugger 0.10.0+dev-gf0767a31 (2018-06-11-13:40)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 0
none separate
adapter speed: 400 kHz
cortex_m reset_config sysresetreq
Error: unable to find CMSIS-DAP device
Error: No Valid JTAG Interface Configured.
Error: No Valid JTAG Interface Configured.
```
### Arduino Zero -> Native port
```
Arduino: 1.8.9 (Linux), Board: "Arduino/Genuino Zero (Native USB Port)"

Build options changed, rebuilding all
Sketch uses 13128 bytes (5%) of program storage space. Maximum is 262144 bytes.
No device found on ttyACM0
An error occurred while uploading the sketch

```
### Arduino Zero -> Programming port
```
Arduino: 1.8.9 (Linux), Board: "Arduino/Genuino Zero (Programming Port)"

Build options changed, rebuilding all
Sketch uses 13128 bytes (5%) of program storage space. Maximum is 262144 bytes.
Open On-Chip Debugger 0.10.0+dev-gf0767a31 (2018-06-11-13:40)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 0
none separate
adapter speed: 400 kHz
cortex_m reset_config sysresetreq
Error: unable to find CMSIS-DAP device
Error: No Valid JTAG Interface Configured.
Error: No Valid JTAG Interface Configured.
```

reillyeon, 2019-09-04:
When you are on the USB internals page for the device click the "Get WebUSB & Microsoft OS 2.0 Descriptors" button. That page as a JavaScript version of the same logic Chrome internally uses to try to read the descriptor and should give you better errors/diagnostic information.

armsp, 2019-09-04:
@reillyeon I get the following for arduino webusb -
```
Offset is outside the bounds of the DataView
Descriptor is too long.
```
For Adafruit TinyUSB webusb i get -
[image]

I am afraid I don't really know what to make of this information.

jpliew, 2019-09-04:
@armsp when you use ***Adafruit Feather M0 Express*** board with an edited `USBCore.h`

[image]

the WebUSB Landing Page is empty. This means, Adafruit's USB must have something that is causing this.

The reason why I asked you to try with Arduino boards is because we need to understand if there are any differences between Arduino and Adafruit's USB stack, from a quick look at their source files, I can see they are different. Instead of finding out the differences, one quickest way is to test with Arduino boards by changing Arduino's `USBCore.h` and use Arduino boards, in you case ***Arduino MKRZERO*** is working and I suggest to test with this ***Arduino MKRZERO*** board first.

Once you have edited the `USBCore.h` for ***Arduino MKRZERO*** then check with chrome's debug, check if USB version and WebUSB landing page is there.

This is just a simple comparison debugging technique, once you get one board working, it will be more easier and confident for you to tackle Adafruit's board files.

armsp, 2019-09-05:
@jpliew I agree.
So when I try with _Arduino MKRZERO_ with the changes in `USBCore.h`, I **don't** see any notification AND the bad news is that it is **not even enumerated** in ```chrome://usb-internals```

When I try just with _Arduino MKRZERO_ WITHOUT any changes to `USBCore.h` then **it is enumerated** in `chrome://usb-internals`. It pairs too but when I actually try to connect and run the `console` example I get this error -
```
Connecting to Arduino MKRZero...
Connection error: SecurityError: Access denied.
```
The **Get WebUSB & Microsoft OS 2.0 Descriptors** button shows this -
```Failed to read the device BOS descriptor to determine the total descriptor length. (Reason: Transfer Error)```

And the rest of the details are as follows (still NO landing page) -
[image]

jpliew, 2019-09-05:
@armsp there are two `USBCore.h` in Arduino. One in

`arduino/avr/cores/arduino/USBCore.h`

and the other one in 

`Arduino15\packages\arduino\hardware\samd\1.8.3\cores\arduino\USB`

Have you try editing both files?

I don't have ***Feather M0 Express*** with me, once I get one I will test this out for you.

armsp, 2019-09-05:
@jpliew 
The `USBCore.h` at `~/arduino-1.8.9/hardware/arduino/avr/cores/arduino` has been changed permanently and as thats what the README said. I don't revert it back at all.

The experiments I have been running are with the following files -
* Arduino's samd core : `~/.arduino15/packages/arduino/hardware/samd/1.8.3/cores/arduino/USB`
* Adafruit's samd core : `~/arduino15/packages/adafruit/hardware/samd/1.5.3/cores/arduino/USB`

_I just wanted to collate all the experiments that I have done so far. Apparently things are not what they seem._

**When Adafruit Feather M0 is selected as Board, then I make changes to `USBCore.h` in Adafruit's samd core location as my understanding is that while compiling that will be used**
 
| Board | Library | `USBCore.h` version | USB Version | Landing Page shows? | Notification | Does it work |
| --- | --- | --- | --- | --- | --- | --- |
| Adafruit Feather M0 Express | Adafruit Tiny USB - webusb | 0x200 | 32.1.0 | Yes | Yes | Yes |
| Adafruit Feather M0 Express | Adafruit Tiny USB - webusb | 0x210 | 32.1.0 | Yes | Yes | Yes |
| Adafruit Feather M0 Express | arduino  webusb | 0x200 | 2.0.0 | No | No | Yes |
| Adafruit Feather M0 Express | arduino  webusb | 0x210 | 2.1.0 | No | No | Yes|

**When Arduino MKRZERO is selected as Board, then I make changes to `USBCore.h` in Arduino's samd core location as my understanding is that while compiling that will be used**

| Board | Library | `USBCore.h` version | USB Version | Landing Page shows? | Notification | Does it work |
| --- | --- | --- | --- | --- | --- | --- |
| Arduino MKRZERO | webusb | 0x210 | NA | NA | No | NA |
| Arduino MKRZERO | webusb | 0x200 | 2.0.0 | No | No | No |

### Takeaways
* Arduino's `USBCore.h` location : `~/.arduino15/packages/arduino/hardware/samd/1.8.3/cores/arduino/USB`
* Adafruit's `USBCore.h` location  : `~/arduino15/packages/adafruit/hardware/samd/1.5.3/cores/arduino/USB`
 
Changes to the `USBCore.h` USB version don't actually affect the USB Version for Adafruit TinyUSB library and moreover it stays at **32**.1.0 . This means there is some other file/value that takes precedence. Or we are missing the bigger picture here.