webusb/arduino #70: Chrome Version and Android

puttley, 2019-10-21:
Hello - I have a webUSB device that works perfectly with the latest versions of Chrome on Win 10 and Mac OS. However, the newest version of Chrome for Android will no longer work...earlier versions did.

When attempting to connect using an Android device, I get a SecurityError: Access denied message. My device will show up in the device selection window, but when I select it and attempt to connect I get a NotFoundError: No device selected.

Is there a known issue with Android and the latest version for Chrome 77.0.3865.116 ?

reillyeon, 2019-10-21:
This is a known issue with Chrome 77 and has been fixed in Chrome 78 ([issue 1009362](https://bugs.chromium.org/p/chromium/issues/detail?id=1009362)). You can verify this issue has been fixed by installing [Chrome Beta](https://play.google.com/store/apps/details?id=com.chrome.beta) from the Google Play Store. According to [the schedule](https://chromiumdash.appspot.com/schedule) Chrome 78 should be released to stable-channel users shortly.

closed 2019-10-21 by reillyeon

puttley, 2019-10-21:
Thanks for the prompt reply. I downloaded and installed Chrome Beta 78.0.3904.62. The webUSB device appears in the device list window, but once I select it and attempt to connect, Chrome crashes and closes. Still, no issues with the latest versions of Chrome on Win 10, Mac and Chromebook - only Android.

reillyeon, 2019-10-22:
I've filed an issue internally to track this. Thanks for checking.

puttley, 2019-10-22:
Very much appreciated,  thank you.

puttley, 2019-11-01:
Has there been an update on this issue?

reillyeon, 2019-11-01:
The issue has been fixed in Chrome 78.0.3904.90 and is rolling out to stable-channel now.

puttley, 2019-11-02:
Good news. Thank you.