webusb/arduino #73: Nano 33 iot vs BLE

j3nsykes, 2020-03-04:
I see the Arduino Nano 33 iot listed as a compatible board. Is  there a reason the 33 Nano BLE is not ? I cannot get it  to compile with the latter board but wondering if there is a change  to a .h file I can make which would make it compile.

reillyeon, 2020-03-04:
What is the compiler error you are getting?

j3nsykes, 2020-03-04:
Arduino: 1.8.10 (Mac OS X), TD: 1.49, Board: "Arduino Nano 33 BLE"

Multiple libraries were found for "WebUSB.h"
In file included from /Users/j.sykes/Documents/Arduino/webUSB_send/webUSB_send.ino:1:0:
 Used: /Users/j.sykes/Documents/Arduino/libraries/WebUSB
/Users/j.sykes/Documents/Arduino/libraries/WebUSB/WebUSB.h:27:10: fatal error: PluggableUSB.h: No such file or directory
 #include "PluggableUSB.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board Arduino Nano 33 BLE.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

I do not have multiple WebUSB libraries installed incidentally.

reillyeon, 2020-03-04:
According to the tech specs the Nano 33 BLE has native USB support in the nRF52840 chip while the Nano 33 IoT uses a SAMD21 chip. It doesn't look like the PluggableUSB library is supported on the Nordic chip yet.

j3nsykes, 2020-03-04:
no problem thanks!