webusb/arduino #74: "No device found" for unlisted atmega32u4 board

leanofis-iot, 2020-03-19:
Hi,

I am trying to use webusb with another atmega32u4 board.

Can you help me ?

Sparkfun Pro Micro 3.3V 8MHz board.
Windows 10
Chrome 80.0.3987.149

The issue is "No device found" and no connection.

Installation steps;

1. Installed WebUSB arduino library. I didn't transfer webusb/avr hardware package to arduino hardware folder yet.
2. Copied website files to my Github Page
3. demos > serial.js, in filters ADDED LINE: { 'vendorId': 0x1B4F, 'productId': 0x9204 }, // Sparkfun Pro Micro 3.3V 8MHZ 
		Tried also only Vendor ID, { 'vendorId': 0x1B4F }, // Sparkfun
4. ..avr > core > arduino > USBCore.h, MODIFIED LINE #130: #define USB_VERSION 0x210
5. Arduino IDE, selected Sparkfun Pro Micro 3.3V 8MHz board and uploaded RGB demo sketch to board.

Device manager here;

[image]

reillyeon, 2020-03-19:
From the exclamation point on the "SparkFun Pro Micro" entry in Device Manager screenshot it looks like Windows is not recognizing the descriptors that this library adds and automatically loading the right driver. The device's other interface, the USB serial device, is properly recognized and has the right driver loaded. Can you look at the properties for the broken device and show us what error Windows reports?

leanofis-iot, 2020-03-19:
hi @reillyeon

Now the issue is, SecurityError: Access Denied

Showing by Zadig software, for Sparkfun Pro Micro board, 2 different device drivers;

interface 0 ( usbser )

[image]

interface 2 ( no driver ) 
Should install winusb driver here ?

[image]

events for the broken device;

Device USB\VID_1B4F&PID_9204&MI_02\7&28d8bbf4&0&0002 was configured.
Driver Name: null
Class Guid: {00000000-0000-0000-0000-000000000000}
Driver Date: 
Driver Version: 
Driver Provider: 
Driver Section: 
Driver Rank: 0x0
Matching Device Id: 
Outranked Drivers: 
Device Updated: false
Parent Device: USB\VID_1B4F&PID_9204\WUART  

chrome errors; 
1. error when page loaded, 
2. error when clicked the connect button

[image]

reillyeon, 2020-03-19:
Using the Zadig tool to force the WinUSB driver to load will work around the problem I described in my last comment. The Microsoft OS 2.0 descriptors that this library adds to the device should be telling Windows to load that driver by default.

Recent builds of Chrome have a tool for debugging these issues but you will need to run it on a platform other than Windows, or use the Zadig tool first so that you can overcome the "access denied" error. This tool mimics the steps that Windows and Chrome perform when enumerating a new USB device but display the results in a way that can help to debug issues.

1. Visit chrome://usb-internals.
2. Click on the "Devices" tab.
3. Find your device in the list and click its "Inspect" button.
4. Click the "Show WebUSB & Microsoft OS 2.0 Descriptors" button.
5. Expand the "Microsoft OS 2.0 Descriptor" and "Microsoft OS 2.0 Descriptor Set Information" nodes and click the "GET" button next to the vendor code.
6. Expand the "Microsoft OS 2.0 Compatible ID Descriptor" node.
7. The "Compatible ID string" should be equal to "WINUSB". If any of these descriptors fail to load (the tool is reading them directly from the USB device as you click the "get" buttons) that means that there is a bug in the library and your device isn't responding correctly. This would explain why Windows is not behaving as expected.

Screenshot:
[image]

leanofis-iot, 2020-03-20:
I tried zadig winusb install, but the issue still remains.
Now, reserved windows.

I tested on Android OK, a successful communication between arduino and chrome.

Do webusb support readline parser with newline delimiters ?

reillyeon, 2020-03-20:
> I tried zadig winusb install, but the issue still remains.

Have you tried the debugging steps I suggested?

> Do webusb support readline parser with newline delimiters ?

WebUSB provides the ability to read data from the device using USB transfers. Parsing that data is the responsibility of your application.

leanofis-iot, 2020-03-20:
There is a problem with get descriptors;

[image]

[image]

reillyeon, 2020-03-20:
As a sanity check can you make sure that the device or configuration descriptors can be read correctly. If they can then there is a bug in this library which needs to be investigated.

leanofis-iot, 2020-03-20:
Really sorry but I am not good at programming. There must be a sequential error in my code when calling; request/get/open of device and port. Do you think so ?

Here I would ask if WCID driver is useful.

[image]

reillyeon, 2020-03-20:
If you think there is an error in your code please post it here.

The normal "install driver" option is fine. Please confirm that when you've installed the driver you no longer get an "access denied" error. Please post a screenshot of the chrome://usb-internals tool where you have clicked the "Get Device Descriptor" button after installing the WinUSB driver.

leanofis-iot, 2020-03-20:
Firstly please excuse for I keep you busy.

Winusb driver installed by Zadig.

And this flow:

When opened page; 
SecurityError: Access Denied

After clicked "Connect" button to force reconnect; 
Pop-up appears
select device and done paired 
SecurityError: Access Denied

Here usb internals for all the get are clicked:

[image]

leanofis-iot, 2020-03-20:
Need to set in chrome flags ?

reillyeon, 2020-03-20:
No Chrome flags are necessary. Since this device is working for you on Android please try these steps in chrome://usb-internals on your Android device to compare.

leanofis-iot, 2020-03-21:
@reillyeon 

IT WORKS !!

Earlier I played with chrome flags.
Now, restored to DEFAULT.

Windows Chrome;
Connect, send, receive, data communication is OK.

Thank you very much.

closed 2020-03-21 by reillyeon