webusb/arduino #75: Bricked MKRZero

lbrutti, 2020-04-22:
Hello everyone,
I bricked my MKRZERO by setting 
`#define USB_VERSION 0x200`

to 0x20**1** instead of **0x210**

Anybody knows how to debrick it?

I have an Arduino Uno, MKR1400 and MKR1300: can I use one of them to reflash the bootloader?

thanks

reillyeon, 2020-04-22:
What host OS are you running? For the most part macOS and Linux are much more flexible about ignoring weird USB versions.

lbrutti, 2020-04-23:
Windows 10, but I also have a MacBook: do you think that loading a sketch
from osx will unbrick the board?

ubidefeo, 2020-04-23:
@lbrutti 
if the IDE sees your board (although the VID/PID) won't let it automatically recognise it, 
just manually select the MKRZERO as a target and the serial port and flash a blink sketch.
Every time you program one of the MKR boards, the VID/PID pair is reflashed

ubidefeo, 2020-04-23:
@lbrutti I think I misunderstood the issue there

lbrutti, 2020-04-23:
> 
> 
> What host OS are you running? For the most part macOS and Linux are much more flexible about ignoring weird USB versions.

Unfortunately neither OSx sees the board as attached to serial port.
Is there a procedure to burn the bootloader again as in https://www.instructables.com/id/How-to-fix-bad-Chinese-Arduino-clones/ ?

ubidefeo, 2020-04-23:
@lbrutti 
you can burn the bootloader only if you use an openOCD compatible programmer.
Atmel ICE
BlackMagic Probe
J-Link

lbrutti, 2020-04-23:
I managed to recover the board by pressing reset twice...I wasn't double pressing fast enough.

Thanks to all!

closed 2020-04-23 by lbrutti