webusb/arduino #78: ESP32 and ESP8266 support

TedNIVAN, 2020-11-10:
Hi,

Is there a way to make this work with **ESP32** and **ESP8266** boards?

Kind regards,

jpliew, 2020-11-10:
@TedNIVAN both MCU have no native USB support.

TedNIVAN, 2020-11-10:
@jpliew That's correct.
However I have an ESP32 development board using the **CP2102 USB to UART Bridge Controller**.

I've added the VID and PID to the filters:

    const filters = [
      { 'vendorId': 0x2341, 'productId': 0x8036 }, // Arduino Leonardo
      { 'vendorId': 0x2341, 'productId': 0x8037 }, // Arduino Micro
      { 'vendorId': 0x2341, 'productId': 0x804d }, // Arduino/Genuino Zero
      { 'vendorId': 0x2341, 'productId': 0x804e }, // Arduino/Genuino MKR1000
      { 'vendorId': 0x2341, 'productId': 0x804f }, // Arduino MKRZERO
      { 'vendorId': 0x2341, 'productId': 0x8050 }, // Arduino MKR FOX 1200
      { 'vendorId': 0x2341, 'productId': 0x8052 }, // Arduino MKR GSM 1400
      { 'vendorId': 0x2341, 'productId': 0x8053 }, // Arduino MKR WAN 1300
      { 'vendorId': 0x2341, 'productId': 0x8054 }, // Arduino MKR WiFi 1010
      { 'vendorId': 0x2341, 'productId': 0x8055 }, // Arduino MKR NB 1500
      { 'vendorId': 0x2341, 'productId': 0x8056 }, // Arduino MKR Vidor 4000
      { 'vendorId': 0x2341, 'productId': 0x8057 }, // Arduino NANO 33 IoT
      { 'vendorId': 0x1A86, 'productId': 0x7523 }, // ESP8266
      { 'vendorId': 0x10C4, 'productId': 0xEA60 }, // ESP32
      { 'vendorId': 0x239A }, // Adafruit Boards!
    ];

I can detect the board but nothing happens when I try to connect.

**Do you have any idea how can I make this work?**

jpliew, 2020-11-11:
CP2102 is a USB to TTL bridge. It's work is just to translate USB to TTL, it will not work for WebUSB. By just adding the VID/PID will not make it work. 

If you really want to make this work, you can use ATmega32U4 (Arduino Leonardo) acting as a WebUSB converter and then connecting it to ESP32 / ESP8266 via UART.

But then since you have a wireless chip, why would you want to have it connected to the web using a wire/cable? 

If you really want to connect the ESP to the web browser, maybe consider Web Bluetooth with ESP32 ? 
https://github.com/kpatel122/ESP32-Web-Bluetooth-Terminal

Hope this helps.

TedNIVAN, 2020-11-16:
Hi @jpliew 
Thanks for your explanation. It makes lots of sense.

The WiFi is already taken for IoT purposes on the ESP32. So I can't use Bluetooth.
I want to use the USB port to get some device information.

**1. Is there another library that could work with a USB TTL Bridge over the web?
2. Maybe I should use the ESP32-S2 which seems to have WebUSB support instead of the ESP32?**

jpliew, 2020-11-16:
@TedNIVAN

As I said, you can replace the CP2102 with ATmega32U4. ATmega32U4 can perform 2 functions, USB to TTL and also WebUSB if you know what you are doing.

Alternatively like you said, ESP32-S2, but not sure how much work is needed to make it work. You can ask `chegawara` how this was perform https://esp32.com/viewtopic.php?t=15981

jpliew, 2020-11-17:
Coming back to what you are working on, is there anything that the ESP32 using WIFI cannot do with the browser that the WebUSB can do?

TedNIVAN, 2020-11-17:
The WIFI is already used for another application. I'll go for the ESP32-S2.
Thanks for your time.

jpliew, 2020-12-30:
@TedNIVAN please close this issue if you think that your questions were answered. Thanks.

thujer, 2020-12-30:
Hi, I'm working on ESP32 device with WebUSB to Javascript communication.

We decide to use ESP32 with Arduino Leonardo combination, because Leonardo
has native USB support.

On the way to our solution I found a few projects about trying to bring
software USB support to ESP32.

I have not direct experience, but I think, that it can be what You looking
for.

Here are a few links which I found

https://github.com/cnlohr/espusb
https://www.electronics-lab.com/espusb-usb-software-stack-esp8266/

jpliew, 2020-12-30:
In my opinion @thujer method, ESP + ATmega32U4 is the best, quickest, easiest and cheapest to get ESP the WebUSB features. 

But I am intrigue why one would want a wireless ESP that can already talk to web app natively to be bound to wire just to let it talk to web app.

thujer, 2020-12-30:
@jpliew Yes, its ESP32 connected to WebUSB via Arduino is easiest solution.

Our project is client online and offline application which sending user data from ESP32 device to javascript to visualization on graph. We need to let user to able send their data to server. Thats why we cannot use WifiAP mode.

I had thoughts about a solution with a USB serial and an application that would work as a local server, but it would no longer be a pure web application and it would be necessary to install the application on a PC. We wanted to avoid that.

Our device is connected to next mobile device via ESPNow.

I'm not proud of a few communications, but we have a working solution for now, before we come up with something better.

Web browser <-WebUSB-> Arduino Leonardo <-Serial-> ESP32 <-ESPNow-> ESP32 (mobile device)

reillyeon, 2020-12-30:
I could imagine that for initial configuration connecting the device physically would be useful. Afterwards all communication would happen wirelessly.