webusb/arduino #79: WebUSB on Android Smartphone

TedNIVAN, 2020-11-16:
Hi there,

I have an app (link [here](https://tednivan.github.io/fruity-market/app.html)) using webusb that works well on PC. I'm able to detect an Arduino board and connect to it.
However when I try to use the same app on an Android phone (Android version 10), I can detect the device but nothing happens when I click the connect button.

**1. Do you guys know how to make a webusb app work on Android?
2. Maybe there are some permissions/settings to enable on the phone?**

Kind regards,

reillyeon, 2020-11-16:
Which board are you using? There's a known issue (sorry, I can't find the link right now) where SAMD boards don't respond to descriptor requests properly and that breaks connections on Android.

TedNIVAN, 2020-11-17:
I'm using the **Arduino NANO 33 IoT** board which is indeed a SAMD board.

**Any workaround?**