webusb/arduino #84: MRKZero support?

Delonous, 2021-02-03:
Tried compiling for a MRK Zero and get this. I'm running a fresh install of Arduino IDE 1.8.3. All recently updated libs, can get versions if needed.

`\Arduino\libraries\WebUSB/WebUSB.h:25:10: fatal error: USB/PluggableUSB.h: No such file or directory
 #include "USB/PluggableUSB.h"
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board Arduino MKRZERO.`

I then tried compiling with Leonardo and it worked fine. Am I missing something?

jpliew, 2021-02-03:
# Problem
The path for `PluggableUSB.h` has been changed since 1.8.10 of SAMD board. Hope this helps. 

# Fix
Go to Arduino board manager, choose Arduino SAMD boards by Arduino, downgrade the version to 1.8.9

Delonous, 2021-02-03:
Thanks for the tip. Solved the issue.

closed 2021-02-03 by Delonous