webusb/arduino #86: WebUSB requires a board that supports USB client device mode

foresthan, 2021-03-02:
I got the following error while uploading the sketch.ino in rgb/sketch folder. 

Documents\Arduino\libraries\WebUSB/WebUSB.h:33:2: error: #error "WebUSB requires a board that supports USB client device mode."
 #error "WebUSB requires a board that supports USB client device mode."
  ^~~~~
exit status 1
Error compiling for board Arduino Uno.

The device I am connecting is "Adafruit Circuit Playground Express" (VendorID = 0x239A). Does it not support WebUSB?

Thanks in advance!

reillyeon, 2021-03-02:
I believe that this library will work on that board but you need to configure the Arduino IDE properly for programming it by following the instructions here: https://learn.adafruit.com/adafruit-circuit-playground-express/set-up-arduino-ide

That board is not an Arduino Uno.

closed 2021-03-02 by reillyeon

foresthan, 2021-03-02:
Thanks for the reply. Yes, it works now.  But when I configurate the IDE I got another compiling failure:
fatal error: USB/PluggableUSB.h: No such file or directory
The root cause of this error is not in this project, but in the Arduino SAMD board package 1.8.10 and newer. 

Just in case anyone faces the same issue I had, the solution is to downgrading Arduino SAMD board package to 1.8.9