webusb/arduino #89: Question about usage of controlTransferOut() at connect() and disconnect() in serial.js

Kazhuu, 2021-07-10:
Hi and thanks for this awesome project. I'm currently learning to use WebUSB and this repo is really a gold mine. I've been looking through the code and saw `controlTransferOut()` in `serial.js` at `connect()` and `disconnect()`. Like [here](https://github.com/webusb/arduino/blob/42c7bcde01a0e52218b068c87238867abed6d436/demos/serial.js#L77). And kept wondering what are the `request` and `value` numbers here exactly in this context?

I read found that they are vendor specific. So then I looked through WebUSB code on arduino side but could't find anything resembling these values from there. Could you please shed some light what are these and are they required? Event point to further reading would be helpful! Thanks you a lot!

jpliew, 2021-07-11:
Hi @Kazhuu this link will provide you more information https://www.beyondlogic.org/usbnutshell/usb6.shtml

Kazhuu, 2021-07-11:
This looks like a good resource to read. Thank you a lot!

reillyeon, 2021-07-12:
The code which handles these SET_CONTROL_LINE_STATE requests is here: https://github.com/webusb/arduino/blob/gh-pages/library/WebUSB/WebUSB.cpp#L242

commit 407600fc (2021-07-12): Explain SET_CONTROL_LINE_STATE requests

This change adds comments explaining the use of the USB CDC-ACM SET_CONTROL_LINE_STATE requests on the vendor-specific interface provided by the WebUSB Arduino library.

Fixes #89.
closed 2021-07-12 by reillyeon