webusb/arduino #91: Not working with Seeeduino XIAO

ratheeshpv, 2021-11-22:
I followed all instructions as in https://github.com/webusb/arduino, to work WebUSB with my 'Seeeduino XIAO' board,
Modified landing address to, WebUSB WebUSBSerial(1, "localhost:8888");
I can upload the sketch by using board 'Seeeduino femto' and 'Arduino USB' 
Opened the html after adding VID and PID, on local like, "file:///C:/Users/......."
But showing "No device found"
In device manager 'Seeeduino fermto M0' driver was not installed, So I installed 'Zadic' to install the driver to 'WinUSB (V6.17600...)', but the result is same
I am using windows 7
chrome is  96.0.4664.45 (64bit)
Arduino 1.8.13
Please help anybody to solve this issue,

Thanks in advance

jpliew, 2021-11-23:
Hi,

If  your board is not detected, something is wrong with the installation of webusb. Zadic will not help. You need to uninstall whatever Zadic has installed for you.

If you are using Xiao, it will be easier to use TinyUSB with WebUSB.

First you need to install the driver from Seeed
https://github.com/Seeed-Studio/Signed_USB_Serial_Driver

Then Read how to use Tinyusb for Xiao
https://www.seeedstudio.com/blog/2020/04/30/tinyusb-stack-on-seeeduino-xiao/

Then read how to use WebUSB in TinyUSB
https://learn.adafruit.com/using-webusb-with-arduino-and-tinyusb?view=all

Hope this helps.

ratheeshpv, 2021-11-23:
> 
@jpliew, Thanks , I will try as you advised.

ratheeshpv, 2021-11-26:
@jpliew , I went through all steps you advised and tried webUSB example form  https://learn.adafruit.com/using-webusb-with-arduino-and-tinyusb?view=all 
but I cannot upload the program in to Seeeduino XIAO, when I asked they told, this board is not supported need to ask in their forum
I didn't get any positive reply from seeed studio.
Can you help me to choose another board (similar to seeeduino XIAO which work with https://github.com/webusb/arduino webusb example ) ?

jpliew, 2021-11-26:
@ratheeshpv from your description here https://github.com/adafruit/Adafruit_TinyUSB_Arduino/issues/140 I think you have an installation problem. The error message is not upload problem, it was compilation problem. The XIAO board should be able to use the library properly, they just not willing to support board bought from others.

I did the same and I was able to compile webusb_rgb properly

[image]

I would suggest you try on another computer follow this steps

1. Install Arduino IDE and also XIAO driver on Windows
2. Install XIAO board json file https://files.seeedstudio.com/arduino/package_seeeduino_boards_index.json
3. Try to compile one example and make sure that XIAO is working properly. Try Blink example and make sure XIAO LED can blink.

DO NOT PROCEED until you get the above working.

4. Install Adafruit TinyUSB library https://github.com/adafruit/Adafruit_TinyUSB_Arduino/archive/refs/heads/master.zip
5. Click File->Example->Adafruit TinyUSB Library->WebUSB->webusb_rgb
6. Click the tick sign 
7.

ratheeshpv, 2021-11-26:
@jpliew Thank you for your help, I will try this and let you know.

ratheeshpv, 2021-11-30:
Hi @jpliew 
I tried it on another Windows 10 computer, but I am getting same error, what I done is explained below,
1. installed new version of Arduino IDE 1.8.16
2. Downloaded tinyusb from 'https://github.com/adafruit/Adafruit_TinyUSB_Arduino'
3. From Arduino, sketch->Include Library->Add .ZIP Library (selected above downloaded Zip file)
4. Opened File->Examples->Adafruit Library->WebUSB->webusb_rgb
5. pasted 'https://files.seeedstudio.com/arduino/package_seeeduino_boards_index.json'  into 'Additional Boards Manager URLs' (in File->Preference) 
6. from Tools->Boards->Boards Manager, searched for "Seeeduino XIAO" and installed 'Seeed SAMD boards'
7. selected 'Seeeduino XIAO' from Tools->Boards->Seeed SAMD boards->Seeeduino XIAO
8. selected 'Tiny USB' from Tools->USB stacks
9. Then clicked 'Verify' (tick)
10. it gives error 'no such file Adafruit_NeoPixel.h'
11. Then downloaded Neopixel from 'https://learn.adafruit.com/adafruit-neopixel-uberguide/arduino-library-installation'
12. Included in the library using Sketch->Include Library->Add .ZIP Library
13. Again clicked 'Verify' (tick)
14. THen getting error 'cannot declare variable web_usb.....
[image]

jpliew, 2021-11-30:
Did you follow my steps 1 - 3 and get XIAO working first? What result did you get for step 3?

[image]

ratheeshpv, 2021-11-30:
Hi @jpliew
Sorry to miss that information,
In my computer, I upgraded to windows 10 from windows 7
Followed upto step 3, the XIAO board Yellow LED was blinking
then I done 4 to 7, I was getting error 'cannot declare variable web_usb...

In another computer (my brother's ) , didn't do step 3 because, he doesn't have board,
all other he done, at last getting 'cannot declare variable web_usb...' error
Do I try including step 3?

Note: the XIAO board I was installing from 'Boards Manager' (because I couldn't get with with JSON). I think all others I done as you said

Thanks

jpliew, 2021-11-30:
Can you please goto File, preferences, tick Compilation

[image]

Recompile again, and copy and paste all the messages here.

ratheeshpv, 2021-11-30:
@jpliew here is the messages,
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.51.0_x86__mdqgnx93n4wtt\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.51.0_x86__mdqgnx93n4wtt\hardware -hardware C:\Users\sanujadithya\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.51.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.51.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\sanujadithya\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.51.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\sanujadithya\Documents\Arduino\libraries\Adafruit_TinyUSB_Arduino-master\examples\WebUSB\webusb_rgb\libraries -fqbn=Seeeduino:samd:seeed_XIAO_m0:usbstack=tinyusb,debug=off -ide-version=10816 -build-path C:\Users\SANUJA~1\AppData\Local\Temp\arduino_build_656654 -warnings=none -build-cache C:\Users\SANUJA~1\AppData\Local\Temp\arduino_cache_252506 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\sanujadithya\Documents\ArduinoData\packages\Seeeduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Users\sanujadithya\Documents\ArduinoData\packages\Seeeduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.bossac.path=C:\Users\sanujadithya\Documents\ArduinoData\packages\Seeeduino\tools\bossac\1.7.0-arduino3 -prefs=runtime.tools.bossac-1.8.0-48-gb176eee.path=C:\Users\sanujadithya\Documents\ArduinoData\packages\Seeeduino\tools\bossac\1.8.0-48-gb176eee -prefs=runtime.tools.openocd.path=C:\Users\sanujadithya\Documents\ArduinoData\packages\arduino\tools\openocd\0.10.0-arduino7 -prefs=runtime.tools.openocd-0.10.0-arduino7.path=C:\Users\sanujadithya\Documents\ArduinoData\packages\arduino\tools\openocd\0.10.0-arduino7 -prefs=runtime.tools.bossac-1.7.0-arduino3.path=C:\Users\sanujadithya\Documents\ArduinoData\packages\Seeeduino\tools\bossac\1.7.0-arduino3 -prefs=runtime.tools.CMSIS.path=C:\Users\sanujadithya\Documents\ArduinoData\packages\Seeeduino\tools\CMSIS\5.7.0 -prefs=runtime.tools.CMSIS-5.7.0.path=C:\Users\sanujadithya\Documents\ArduinoData\packages\Seeeduino\tools\CMSIS\5.7.0 -prefs=runtime.tools.CMSIS-Atmel.path=C:\Users\sanujadithya\Documents\ArduinoData\packages\Seeeduino\tools\CMSIS-Atmel\1.2.1 -prefs=runtime.tools.CMSIS-Atmel-1.2.1.path=C:\Users\sanujadithya\Documents\ArduinoData\packages\Seeeduino\tools\CMSIS-Atmel\1.2.1 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\sanujadithya\Documents\ArduinoData\packages\Seeeduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=C:\Users\sanujadithya\Documents\ArduinoData\packages\Seeeduino\tools\arm-none-eabi-gcc\7-2017q4 -verbose C:\Users\sanujadithya\Documents\Arduino\libraries\Adafruit_TinyUSB_Arduino-master\examples\WebUSB\webusb_rgb\libraries\Adafruit_TinyUSB_Arduino-master\examples\WebUSB\webusb_rgb\webusb_rgb.ino
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.51.0_x86__mdqgnx93n4wtt\arduino-builder -compile -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.51.0_x86__mdqgnx93n4wtt\hardware -hardware C:\Users\sanujadithya\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.51.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.51.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\sanujadithya\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.51.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\sanujadithya\Documents\Arduino\libraries\Adafruit_TinyUSB_Arduino-master\examples\WebUSB\webusb_rgb\libraries -fqbn=Seeeduino:samd:seeed_XIAO_m0:usbstack=tinyusb,debug=off -ide-version=10816 -build-path C:\Users\SANUJA~1\AppData\Local\Temp\arduino_build_656654 -warnings=none -build-cache C:\Users\SANUJA~1\AppData\Local\Temp\arduino_cache_252506 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\sanujadithya\Documents\ArduinoData\packages\Seeeduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Users\sanujadithya\Documents\ArduinoData\packages\Seeeduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.bossac.path=C:\Users\sanujadithya\Documents\ArduinoData\packages\Seeeduino\tools\bossac\1.7.0-arduino3 -prefs=runtime.tools.bossac-1.8.0-48-gb176eee.path=C:\Users\sanujadithya\Documents\ArduinoData\packages\Seeeduino\tools\bossac\1.8.0-48-gb176eee -prefs=runtime.tools.openocd.path=C:\Users\sanujadithya\Documents\ArduinoData\packages\arduino\tools\openocd\0.10.0-arduino7 -prefs=runtime.tools.openocd-0.10.0-arduino7.path=C:\Users\sanujadithya\Documents\ArduinoData\packages\arduino\tools\openocd\0.10.0-arduino7 -prefs=runtime.tools.bossac-1.7.0-arduino3.path=C:\Users\sanujadithya\Documents\ArduinoData\packages\Seeeduino\tools\bossac\1.7.0-arduino3 -prefs=runtime.tools.CMSIS.path=C:\Users\sanujadithya\Documents\ArduinoData\packages\Seeeduino\tools\CMSIS\5.7.0 -prefs=runtime.tools.CMSIS-5.7.0.path=C:\Users\sanujadithya\Documents\ArduinoData\packages\Seeeduino\tools\CMSIS\5.7.0 -prefs=runtime.tools.CMSIS-Atmel.path=C:\Users\sanujadithya\Documents\ArduinoData\packages\Seeeduino\tools\CMSIS-Atmel\1.2.1 -prefs=runtime.tools.CMSIS-Atmel-1.2.1.path=C:\Users\sanujadithya\Documents\ArduinoData\packages\Seeeduino\tools\CMSIS-Atmel\1.2.1 -prefs=runtime.tools.arm-none- [... cut ...]
Using library Adafruit_TinyUSB_Arduino-master at version 1.7.1 in folder: C:\Users\sanujadithya\Documents\Arduino\libraries\Adafruit_TinyUSB_Arduino-master\examples\WebUSB\webusb_rgb\libraries\Adafruit_TinyUSB_Arduino-master 
Using library Adafruit_NeoPixel-master at version 1.10.0 in folder: C:\Users\sanujadithya\Documents\Arduino\libraries\Adafruit_TinyUSB_Arduino-master\examples\WebUSB\webusb_rgb\libraries\Adafruit_NeoPixel-master 
exit status 1
cannot declare variable 'usb_web' to be of abstract type 'Adafruit_USBD_WebUSB'

thanks

jpliew, 2021-11-30:
Your library for Adafruit_TinyUSB is corrupted, please see line 1 from your debug

```C:\Users\sanujadithya\Documents\Arduino\libraries\Adafruit_TinyUSB_Arduino-master\examples\WebUSB\webusb_rgb\libraries\Adafruit_TinyUSB_Arduino-master\examples\WebUSB\webusb_rgb\webusb_rgb.ino```

It should be 

```C:\Users\sanujadithya\Documents\Arduino\libraries\Adafruit_TinyUSB_Arduino-master\examples\WebUSB\webusb_rgb\webusb_rgb.ino```

You should

1. Uninstall Arduino IDE
2. Backup your folder ` C:\Users\sanujadithya\Documents\Arduino\libraries`
3. Save the backup on a USB or External drive
4. Delete ` C:\Users\sanujadithya\Documents\Arduino\libraries`
5. Download Arduino IDE using the ZIP version (NOT THE WINDOWS APP VERSION) https://www.arduino.cc/download_handler.php?f=/arduino-1.8.13-windows.zip
6. Unzip the Arduino IDE 
7. Run Arduino IDE from the unzipped folder

Then follow my previous steps and report your result.

ratheeshpv, 2021-12-01:
Hi @jpliew 
I done following steps
- Uninstalled Arduino IDE
- Backup folder C:\Users\sanujadithya\Documents\Arduino\libraries
- saved the backup on USB
- deleted C:\Users\sanujadithya\Documents\Arduino\libraries
- Downloaded Zip version, Arduino 1.8.16
- Unzipped
- Run Arduino IDE from unzipped folder
- Installed XIAO board
- uploaded 'Blink' program from example, uploaded successfully and the board is working
- installed Adafruit_TinyUSB library from https://github.com/adafruit/Adafruit_TinyUSB_Arduino/archive/refs/heads/master.zip
- installed Adafruit_Neopixel library
- Clicked File->Example->Adafruit TinyUSB Library->WebUSB->webusb_rgb
- selected the board, 'seeeduino XIAO' and 'TinyUSB' stack
- clicked the tick sign
- below are messages I am getting

-------------------------------------------------------------------
C:\Users\sanujadithya\Downloads\arduino-1.8.16-windows\arduino-1.8.16\arduino-builder -dump-prefs -logger=machine -hardware C:\Users\sanujadithya\Downloads\arduino-1.8.16-windows\arduino-1.8.16\hardware -hardware C:\Users\sanujadithya\AppData\Local\Arduino15\packages -tools C:\Users\sanujadithya\Downloads\arduino-1.8.16-windows\arduino-1.8.16\tools-builder -tools C:\Users\sanujadithya\Downloads\arduino-1.8.16-windows\arduino-1.8.16\hardware\tools\avr -tools C:\Users\sanujadithya\AppData\Local\Arduino15\packages -built-in-libraries C:\Users\sanujadithya\Downloads\arduino-1.8.16-windows\arduino-1.8.16\libraries -libraries C:\Users\sanujadithya\Documents\Arduino\libraries -fqbn=Seeeduino:samd:seeed_XIAO_m0:usbstack=tinyusb,debug=off -vid-pid=2886_802F -ide-version=10816 -build-path C:\Users\SANUJA~1\AppData\Local\Temp\arduino_build_679652 -warnings=none -build-cache C:\Users\SANUJA~1\AppData\Local\Temp\arduino_cache_998947 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.openocd.path=C:\Users\sanujadithya\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino7 -prefs=runtime.tools.openocd-0.10.0-arduino7.path=C:\Users\sanujadithya\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino7 -prefs=runtime.tools.bossac.path=C:\Users\sanujadithya\AppData\Local\Arduino15\packages\Seeeduino\tools\bossac\1.8.0-48-gb176eee -prefs=runtime.tools.bossac-1.7.0-arduino3.path=C:\Users\sanujadithya\AppData\Local\Arduino15\packages\Seeeduino\tools\bossac\1.7.0-arduino3 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\sanujadithya\AppData\Local\Arduino15\packages\Seeeduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=C:\Users\sanujadithya\AppData\Local\Arduino15\packages\Seeeduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.bossac-1.8.0-48-gb176eee.path=C:\Users\sanujadithya\AppData\Local\Arduino15\packages\Seeeduino\tools\bossac\1.8.0-48-gb176eee -prefs=runtime.tools.arduinoOTA.path=C:\Users\sanujadithya\AppData\Local\Arduino15\packages\Seeeduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Users\sanujadithya\AppData\Local\Arduino15\packages\Seeeduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.CMSIS-Atmel.path=C:\Users\sanujadithya\AppData\Local\Arduino15\packages\Seeeduino\tools\CMSIS-Atmel\1.2.1 -prefs=runtime.tools.CMSIS-Atmel-1.2.1.path=C:\Users\sanujadithya\AppData\Local\Arduino15\packages\Seeeduino\tools\CMSIS-Atmel\1.2.1 -prefs=runtime.tools.CMSIS.path=C:\Users\sanujadithya\AppData\Local\Arduino15\packages\Seeeduino\tools\CMSIS\5.7.0 -prefs=runtime.tools.CMSIS-5.7.0.path=C:\Users\sanujadithya\AppData\Local\Arduino15\packages\Seeeduino\tools\CMSIS\5.7.0 -verbose C:\Users\sanujadithya\Documents\Arduino\libraries\Adafruit_TinyUSB_Arduino-master\examples\WebUSB\webusb_rgb\webusb_rgb.ino
C:\Users\sanujadithya\Downloads\arduino-1.8.16-windows\arduino-1.8.16\arduino-builder -compile -logger=machine -hardware C:\Users\sanujadithya\Downloads\arduino-1.8.16-windows\arduino-1.8.16\hardware -hardware C:\Users\sanujadithya\AppData\Local\Arduino15\packages -tools C:\Users\sanujadithya\Downloads\arduino-1.8.16-windows\arduino-1.8.16\tools-builder -tools C:\Users\sanujadithya\Downloads\arduino-1.8.16-windows\arduino-1.8.16\hardware\tools\avr -tools C:\Users\sanujadithya\AppData\Local\Arduino15\packages -built-in-libraries C:\Users\sanujadithya\Downloads\arduino-1.8.16-windows\arduino-1.8.16\libraries -libraries C:\Users\sanujadithya\Documents\Arduino\libraries -fqbn=Seeeduino:samd:seeed_XIAO_m0:usbstack=tinyusb,debug=off -vid-pid=2886_802F -ide-version=10816 -build-path C:\Users\SANUJA~1\AppData\Local\Temp\arduino_build_679652 -warnings=none -build-cache C:\Users\SANUJA~1\AppData\Local\Temp\arduino_cache_998947 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.openocd.path=C:\Users\sanujadithya\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino7 -prefs=runtime.tools.openocd-0.10.0-arduino7.path=C:\Users\sanujadithya\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino7 -prefs=runtime.tools.bossac.path=C:\Users\sanujadithya\AppData\Local\Arduino15\packages\Seeeduino\tools\bossac\1.8.0-48-gb176eee -prefs=runtime.tools.bossac-1.7.0-arduino3.path=C:\Users\sanujadithya\AppData\Local\Arduino15\packages\Seeeduino\tools\bossac\1.7.0-arduino3 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\sanujadithya\AppData\Local\Arduino15\packages\Seeeduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=C:\Users\sanujadithya\AppData\Local\Arduino15\packages\Seeeduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.bossac-1 [... cut ...]
C:\Users\sanujadithya\Documents\Arduino\libraries\Adafruit_TinyUSB_Arduino-master\examples\WebUSB\webusb_rgb\webusb_rgb.ino:77:11: note: suggested alternative: 'USBDevice'
   while( !TinyUSBDevice.mounted() ) delay(1);
           ^~~~~~~~~~~~~
           USBDevice
Using library Adafruit_TinyUSB_Arduino-master at version 1.7.1 in folder: C:\Users\sanujadithya\Documents\Arduino\libraries\Adafruit_TinyUSB_Arduino-master 
Using library Adafruit_NeoPixel-master at version 1.10.0 in folder: C:\Users\sanujadithya\Documents\Arduino\libraries\Adafruit_NeoPixel-master 
exit status 1
cannot declare variable 'usb_web' to be of abstract type 'Adafruit_USBD_WebUSB'

jpliew, 2021-12-01:
Hi @ratheeshpv  please try 

1. Close Arduino IDE
2. Delete `C:\Users\sanujadithya\Documents\Arduino\libraries\Adafruit_TinyUSB_Arduino-master`
3. Run Arduino IDE
4. Click `Sketch -> Include library -> Manage libraries`
5. Type Tinyusb in the search bar

[image]

6. Select version 0.10.5 then click install
7. Click File->Example->Adafruit TinyUSB Library->WebUSB->webusb_rgb
8. Try compile again

I believed Xiao TinyUSB stack has a conflict with the new Adafruit TinyUSB library.

ratheeshpv, 2021-12-02:
Hi @jpliew
Now everything working perfect! THANK YOU SO MUCH!!!! for your help
I tried both examples (rgb and serial), and are working 
May I get your email ID, for any future help regarding webusb or Arduino? my mail ID is 'ratheeshpv29@gmail.com'

Thanks

closed 2021-12-05 by ratheeshpv

Sercurio, 2023-06-20:
Hi,
I have not succeeded to access my Xiao trough web browser despite the information up there.
The sketch compiles but no devices are found in the web browser.

Someone can help ?

My config:
platformio.ini
```
[env:seeed_xiao]
platform = atmelsam
board = seeed_xiao
framework = arduino
lib_deps = 
	; arduino-libraries/MIDIUSB@^1.0.5
	adafruit/Adafruit TinyUSB Library@0.10.5
monitor_speed = 115200
board_build.usb_product = "MIDIPots"
build_flags =
   -DUSE_TINYUSB
lib_archive = no
```

the sketch
```
/*********************************************************************
 Adafruit invests time and resources providing this open source code,
 please support Adafruit and open-source hardware by purchasing
 products from Adafruit!

 MIT license, check LICENSE for more information
 Copyright (c) 2019 Ha Thach for Adafruit Industries
 All text above, and the splash screen below must be included in
 any redistribution
*********************************************************************/

/* This sketch demonstrates WebUSB as web serial with browser with WebUSB support (e.g Chrome).
 * After enumerated successfully, Browser will pop-up notification
 * with URL to landing page, click on it to test
 *  - Click "Connect" and select device, When connected the on-board LED will litted up.
 *  - Any charters received from either webusb/Serial will be echo back to webusb and Serial
 *  
 * Note: 
 * - The WebUSB landing page notification is currently disabled in Chrome 
 * on Windows due to Chromium issue 656702 (https://crbug.com/656702). You have to 
 * go to landing page (below) to test
 * 
 * - On Windows 7 and prior: You need to use Zadig tool to manually bind the 
 * WebUSB interface with the WinUSB driver for Chrome to access. From windows 8 and 10, this
 * is done automatically by firmware.
 */
#include <Arduino.h>
#include "Adafruit_TinyUSB.h"

// USB WebUSB object
Adafruit_USBD_WebUSB usb_web;

// Landing Page: scheme (0: http, 1: https), url
WEBUSB_URL_DEF(landingPage, 1 /*https*/, "adafruit.github.io/Adafruit_TinyUSB_Arduino/examples/webusb-serial/index.html");

int led_pin = LED_BUILTIN;

void line_state_callback(bool connected);

// the setup function runs once when you press reset or power the board
void setup()
{
  pinMode(led_pin, OUTPUT);
  digitalWrite(led_pin, LOW);
  
  usb_web.setLandingPage(&landingPage);
  usb_web.setLineStateCallback(line_state_callback);
  //usb_web.setStringDescriptor("TinyUSB WebUSB");
  usb_web.begin();

  Serial.begin(115200);

  // wait until device mounted
  while( !USBDevice.mounted() ) delay(1);

  Serial.println("TinyUSB WebUSB Serial example");
}

// function to echo to both Serial and WebUSB
void echo_all(char chr)
{
  Serial.write(chr);
  // print extra newline for Serial
  if ( chr == '\r' ) Serial.write('\n');
  
  usb_web.write(chr);
}

void loop()
{
  // from WebUSB to both Serial & webUSB
  if (usb_web.available()) echo_all(usb_web.read());

  // From Serial to both Serial & webUSB
  if (Serial.available())   echo_all(Serial.read());  
}

void line_state_callback(bool connected)
{
  digitalWrite(led_pin, connected);

  if ( connected ) usb_web.println("TinyUSB WebUSB Serial example");
}

```

The build output
```
Processing seeed_xiao (platform: atmelsam; board: seeed_xiao; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelsam/seeed_xiao.html
PLATFORM: Atmel SAM (8.1.0) > Seeeduino XIAO
HARDWARE: SAMD21G18A 48MHz, 32KB RAM, 256KB Flash
DEBUG: Current (atmel-ice) External (atmel-ice, blackmagic, jlink)
PACKAGES: 
 - framework-arduino-samd-seeed @ 1.8.1 
 - framework-cmsis @ 2.50400.181126 (5.4.0) 
 - framework-cmsis-atmel @ 1.2.2 
 - toolchain-gccarmnoneeabi @ 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Adafruit TinyUSB Library @ 0.10.5
Building in release mode
Checking size .pio/build/seeed_xiao/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  12.4% (used 4060 bytes from 32768 bytes)
Flash: [=         ]   6.6% (used 17268 bytes from 262144 bytes)
====================================================================== [SUCCESS] Took 1.44 seconds ======================================================================
 *  Terminal will be reused by tasks, press any key to close it. 
```

The web page 
[image]