webusb/arduino #93: No Connecting Device.

Therealjosephchrzempiec, 2022-01-09:
Hello I'm on windows 10 with the arduino IDE of 1.816. I'm getting Errors saying not connecting When I hit the connect button I'm seeing it shows the board and said Paired but when I try to connect I get these errors.

Connecting to Arduino Leonardo...
Connection error: NotFoundError: No device selected.
Connecting to Arduino Leonardo...
Connection error: InvalidStateError: An operation that changes interface state is in progress.
Connecting to Arduino Leonardo...
Connection error: InvalidStateError: An operation that changes interface state is in progress.

Not sure what to do Can someone please help me?

Joseph

reillyeon, 2022-01-10:
Can you show the code you are running? The `InvalidStateError` means that a call to a method like `claimInterface()` or `selectAlternateInterface()` is still running.

Therealjosephchrzempiec, 2022-01-10:
Hello I'm using the code that is in this Github no changes are made.

Joseph

jpliew, 2022-01-11:
Hi please check this https://github.com/webusb/arduino/issues/44#issuecomment-364642218 and provide the screenshot of

1. Device manager
2. Zadig
3. chrome://device-log
4. chrome://usb-internals