webusb/arduino #96: Feature Request: Arduino PortentaH7

hpssjellis, 2022-06-12:
**Feature Request: Arduino PortentaH7**

@reillyeon any chance of the PortentaH7 being added to this library.  When I plug it into my desktop 
```chrome://device-log/``` says (among other things)

```vid=2341 pid=025B vendor=9025 "Arduino", product=603 "Envie M7",```

If it can't be supported any other options for using webUSB?

reillyeon, 2022-06-13:
If this library doesn't support the PortentaH7 then it either means that this board doesn't support the pluggable USB API (which allows a sketch to add custom USB behavior) or the board-specific API is incompatible.

Now that the Web Serial API is available in Chromium-based browsers this library is not required for sites to access Arduinos. You can use the Web Serial API to access the normal virtual serial port provided by most Arduinos over their USB connection.

hpssjellis, 2022-06-13:
Thanks @reillyeon, great work on this library. Your response  is what I thought but finding it harder than it should be. I have a Seeeduino XIAO which has the new USB support. I would expect the Portenta to also have it.  Any suggestions would be appreciated. I am just using normal serial read and write on the arduino but never seem to get connected and Chrome has nothing useful. This seems to be my issue point

```
    await usbDevice.controlTransferOut({
    requestType: 'class',
    recipient: 'interface',
    request: 0x22,  // vendor-specific request: enable channels
    value: 0x01,  // 0b00010011 (channels 1, 2 and 5)
    index: 2   // Interface 1 is the recipient
  });
```

``` 
  let result = await usbDevice.transferIn(1, 6);   // 1 2 or 5
```

On the Arduino I just have 

```
void setup() {
  Serial.begin(9600);
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop() {
  if (Serial.available() > 0) {
    int serialMessage = Serial.parseInt();
    if(serialMessage == 1) {
      Serial.println("LED On");
      digitalWrite(LED_BUILTIN, LOW);
    } else if(serialMessage == 0) {
      Serial.println("LED Off");
      digitalWrite(LED_BUILTIN, HIGH);
    }
  }
}

```
Which does kind of work with webSerial but slow and only on a few platforms, but no connection with webUSB.

My repo is [here](https://github.com/hpssjellis/my-examples-of-arduino-webUSB-webSerial)

hpssjellis, 2022-06-14:
@reillyeon Yes, I have know idea what I am doing! I am just trying a ton of things to see if I get anything working. So far I can pair a device but never get any full connection. Everywhere I look you seem to be the only expert. 

My github is [here](https://github.com/hpssjellis/my-examples-of-arduino-webUSB-webSerial/tree/main/public)  my active demo site is [here](https://hpssjellis.github.io/my-examples-of-arduino-webUSB-webSerial/public/index.html). Yes it is a mess!

Until someone can easily connect an Arduino to a webpage webUSB seems kind of useless. Today I am going to see what I can do with the [Arduino Create Agent for javascript](https://www.tabnine.com/code/javascript/functions/Agent)

reillyeon, 2022-06-14:
Unfortunately using these hardware APIs requires knowing a little bit about what you're doing because if you don't send a request to the device that makes sense it isn't likely to give you a useful response.

Looking at your code from the previous comment (I'm ignoring your repo because there are too many different code snippets to comment on) the JavaScript is trying to read 6 bytes from the device using `transferIn()` while the Arduino sketch is waiting for the host to send either "1" or "0". That's not going to do anything useful. You want to be calling `transferOut()` and oyu need to figure out which USB endpoint address to use since you are sending data rather than receiving it, check chrome://usb-internals to list your device's endpoints. Similar to the Web Serial samples you need to use a `TextEncoder` to turn a string into a `Uint8Array` you can pass as an argument.

This is probably easier to do with the Web Serial API (and you can use [google/web-serial-polyfill](https://github.com/google/web-serial-polyfill) to support Android devices because you don't need to figure out control transfer parameters and endpoint numbers. I'd like to understand your comment previously about it being "slow".

hpssjellis, 2022-06-14:
@reillyeon 

Loving your [polyfill](https://github.com/google/web-serial-polyfill) I can really work with that. I will be testing the npm [here](https://hpssjellis.github.io/web-serial-polyfill/index.html) 

The basic [demo.html](https://hpssjellis.github.io/web-serial-polyfill/demo.html) runs great on Android, slightly different between my XIAO and PortentaH7. Surprised it does not run on my desktop, do you have a link to an equivalent demo that runs on the desktop?

Great to view the id's after connecting, that is so much easier than having to enter them.
Weird how the vendor and product ID are different than the assigned value.

hpssjellis, 2022-06-15:
@reillyeon I am going to close this issue as the most sensible place to continue this is at the [polyfill](https://github.com/google/web-serial-polyfill) github.

Looks to me like webSerial works fine for desktop computers and webpolyfill for Android. I am getting better speeds now, just need to find a simple webSerial interface.

The new issue on Polyfill is  https://github.com/google/web-serial-polyfill/issues/40

closed 2022-06-15 by hpssjellis

hpssjellis, 2022-06-16:
All done using WebSerial and Web-serial-polyfill

[https://hpssjellis.github.io/web-serial-polyfill/desktop-serial05.html](https://hpssjellis.github.io/web-serial-polyfill/desktop-serial05.html)