webusb/arduino #100: atmega32u4 only connects to Windows not Mac

rykkers, 2023-03-27:
Can i start off by saying, this issue is not caused by this library (although we do use it).

But i need some experts and don't know where else to turn!

We have a custom board based off the atmega32u4, with an external 16MHz crystal oscillator. At present we use the Leonardo bootloader, but have also used the Sparkfun Pro Micro one as well.
The history is that we made a batch of these boards to test them, they all worked perfectly. Connect seamlessly to Windows, ChromeOS & Mac and show up as serial ports.
Then we made a larger batch, now, these new boards connect fine on Windows & ChromeOS, but they don’t connect to Mac.

Thoughts i’ve had, with their counter argument:

- It’s a Mac issue - So why do the old batch, with exactly the same code on them work perfectly on Mac? (I’ve tried flashing an old batch and new batch straight after each other with the same code and only the old batch works)
- It’s a processor problem - But they work perfectly on Windows?
- It’s a USB connector problem - But they work perfectly on Windows?
- Clock speed problem, maybe Windows is more tolerant of clock discrepancies? - Hooked up to an oscilloscope and both the old batch and new batch are reading 16MHz

If anyone has any advice for things to check I'd be most grateful.
Even advice on other forums / areas i could post on.

Many thanks and much love / coffee / beer / cold hard cash to anyone that can point me in the right direction.

jpliew, 2023-03-27:
Hey mate, 

This is probably off topic here. Maybe you can close this issue and send me your schematics (old, new), gerbers (old, new), etc. I will see what do for you.

Cheers
JP

rykkers, 2023-04-11:
Thanks so much, I'll attempt to send you an email / come other form of social

closed 2023-04-11 by rykkers