webusb/arduino #102: how set bouderate? what default speed?

selevo, 2023-09-12:
i not see  it in code ((
thank's

jpliew, 2023-09-13:
The baud rate is set here 

https://github.com/webusb/arduino/blob/486b7e8274b8e39440d6c63719a31273fec60adb/demos/rgb/sketch/sketch.ino#L25

The redefinition of the WebUSB to Serial here 

https://github.com/webusb/arduino/blob/486b7e8274b8e39440d6c63719a31273fec60adb/demos/rgb/sketch/sketch.ino#L13C16-L13C28

is confusing that may be misunderstood as a regular Serial. Should have just let it be 'WebUSBSerial' instead.

reillyeon, 2023-09-13:
Baud rate essentially does not matter for an emulated USB serial port like is created here, which is why there's nothing on the host side to set baud rate.

Now that the [Web Serial API](https://developer.chrome.com/articles/serial/) is available it is mostly unnecessary to use this library to create a custom USB endpoint for WebUSB.

selevo, 2023-09-13:
Thank's!

About API Web Serial ....
It's not work on android brousers
 but webusb - work

https://selevo.github.io/WebUsbSerialTerminal/