webusb/arduino #105: Transfer more than 64ytes

jeremypy972, 2024-10-27:
Hello, I'm trying to transfer JSON between Arduino and Chrome. But the data is divided in 64bytes chunks.
Sounds that Chrome buffer is 4096.

I tired to change the value from 65 to 4096 in transferIn line in serial.js, but it has no effect. Is it something we can configure somewhere ?

Jérémy

jeremypy972, 2024-10-27:
Just after connect, received bytes is more than 64 (75 for instance), but on the second transferin I get data in two times (64+11).

reillyeon, 2024-10-28:
There are two limits, the packet size and the transfer size. The value you pass to `transferIn()` controls the size of the buffer Chrome allocates for the transfer that can be filled by the device. The device sets the packet size in its USB descriptors and can fill that transfer buffer one packet at a time. The device controls when the transfer is "complete" and so it's possible that it only ever responds to a transfer request with 64 bytes.

closed 2024-11-12 by jeremypy972